PubInv/moonrat #6: Design Enclosure with OLED screen

RobertLRead, 2021-04-01:
This is an issue, probably for Harshit, to cover the design of the enclosure for the electronics.  However, it also includes the "bracket" for the battery, and a holder for the thermos jar. We might create sub issues here for these three pieces.

ForrestErickson, 2023-11-07:
@RobertLRead 
@melanielaporte 

I am reopening this to capture enclosure design for the moonratII phase of the project. 
Place enclosure design files at: https://github.com/PubInv/moonrat/tree/main/moonratII/enclosure_mooonratII

## Enclosure Requirements
- [ ] Fit the moonratII printed wiring assembly at mounting points.
- [ ] Opening for the three buttons SW1, SW2 and SW3 with appropriate purchased Key Caps of the vendor of the switches.
- [ ] Opening for the OLED display centered on the display portion
- [ ] Opening for the buzzer.
- [ ] Opening for the power cable
- [ ] Opening for the USB cable
- [ ] Opening and strain relief for the cable to the incubator heater. Note cable might be to a molded connector or a IDC (Insulation displacement connector).
- [ ] 3D printable parts
- [ ] Design in open source CAD software (FreeCAD is preferred as their is already a start ). 
- [ ] Design files for each part and an assembly drawing for the complete enclosure for the enclosure and an assembly drawing for the incubator.

## Reference
The moonratII PCB design files at: https://github.com/PubInv/moonrat/tree/main/moonratII/moonratII
Look for and save manufacturer drawings for purchased parts at: https://github.com/PubInv/moonrat/tree/main/moonratII/moonratII/parts

melanielaporte, 2023-11-12:
Issue #53 has a mechanical drawing for the switch key cap.

closed 2024-02-20 by melanielaporte