PubInv/moonrat #20: Design of 3D printable "puck"

RobertLRead, 2021-06-22:
Note: This should have two loose "chambers" (not enclosed, just framed). The upper should hold the Petrifilms, and the lower should hold the heating cloths. I recommend a "waffle" construction. It should have "legs" which reach down to the bottom of the thermos to keep the other two areas suspended. Contact with the metal of the thermos should be relatively small, and relatively tight (maybe fit with a slight compression of the material so it doesn't rattle. Note that a modification of this puck could hold a full petri dish.

ErickAlex, 2022-01-30:
Hi, Please let me know how I can contribute. I am a design engineer with 10+ years experience. I am also good with metal and woodworking if needed. I am based out of Houston, TX.

RobertLRead, 2022-01-31:
@ErickAlex Thank you. Can you please email me at <read.robert@gmail.com> ?  What we really need right now is an Arduino/electrical engineer for this project. But I would love to use you even if you are only do 3D modeling, and there are several interesting projects that might interest you.

closed 2022-08-08 by RobertLRead