PubInv/moonrat #31: Fundamental Modularity: The 6-wire appliance, Choose Connector

ForrestErickson, 2023-05-18:
Fundamental Modularity: The 6-wire appliance
The electronic elements of this incubator can be separated into two modules. As usual, identifying this modularity allows improvements in our ability to evolve the design. The heat-insulation thermos is a third module.

Call the part that goes into the thermos the "appliance" and the parts that are outside it the "controller". The appliance consists of a heating element and a thermometer, the wires and a connector. The controller consists of a microcontroller, an electronic power circuit, and some means of programming the controller, all packed into an enclosure.

The "appliance" needs only 6 wires:

GND (ground, nominally voltage 0.)
+12V PWR (the heating elements requires approximately 12V power.)
HEAT ON (Vin)
+5V (for the thermometer).
SIG -- data from the digial "one-wire" format
UNK - a final wire is included for future functionality that is unknown right now.
These (esp. the power) should probably be rated for 2 amps, but in practice 1 amp is probably good enough.

We have not yet chosen the connector, but we would like the connector to be of minimal width, so, for example, a 3x2 JST layout would be better than 6 pins in line.

ForrestErickson, 2023-05-18:
Lee reviewed the current rating per pin on RJ45 connectors and found they are only rated at 1.5A per pin so not much margin .  If two pins had +5 and two pins had GND it would have some redundancy.  

Still looking for ideas.

ForrestErickson, 2023-05-20:
Inexpensive, Off the shelf 10 conductor ribbon cable
https://www.amazon.com/gp/product/B01DP13EDA/ref=ox_sc_saved_title_2?smid=A64W1E1ZZHST0&th=1
[image]

What appears to be the mating connector:
https://www.digikey.com/en/products/detail/molex/0702461001/760166
[image]

ForrestErickson, 2023-05-24:
Another idea is the molded connectors used for automotive lights to trailers.

https://www.amazon.com/807-Connectors-Extension-Connector-Way-Socket/dp/B07SR28CG4/ref=sr_1_1

magnetictheory, 2023-05-27:
Image of automotive connector: [807 Square 6-Way Trailer Wiring Harness Connectors,6 pin Square Trailer Wire Extension for LED Brake Tailgate Light Bars, Hitch Light Trailer Wiring Harness Extension Connector (6 Way-Socket)](https://www.amazon.com/807-Connectors-Extension-Connector-Way-Socket/dp/B07SR28CG4/ref=sr_1_1)

[image]

ForrestErickson, 2023-07-28:
@melanielaporte 
See what we found!!!!

At JLCPCB found 2x5 plug
https://jlcpcb.com/partdetail/boomele_boom_Precision_elec-C5665/C5665

[image]

Manufacturer BOOMELE(Boom Precision Elec)
MFR.Part # C5665
JLCPCB Part # C5665

ForrestErickson, 2023-07-30:
Have placed the following in to the PCB:
Manufacturer BOOMELE(Boom Precision Elec)
MFR.Part # C5665
JLCPCB Part # C5665

ForrestErickson, 2023-07-30:
Have placed the following into the PCB:
0.4614	1x6P 20A -40℃~+105℃ 300V Green 12~30 Straight pin 5.08mm 2.5 1 6 Plugin,P=5.08mm Screw terminal ROHS	JLCPCB	C2915642			DB128V-5.08-6P-GN-S	DIBO

ForrestErickson, 2023-11-04:
@melanielaporte 
This may be ready to close out. The PCB has a screw terminal which will work with the automotive connector. The PCB also has the IDC header which is an alternative way to connect too. 

We need to check that we have a BOM only part for the automotive connectors and or the IDC cable.

melanielaporte, 2023-11-12:
Complete. Has been added to the schematic.

melanielaporte, 2023-12-02:
Ordered cable connectors for J9 and J6

https://www.amazon.com/CARROFIX-Trailer-Extension-18-Gauge-Connector/dp/B08B4PBGGR/ref=pd_rhf_sc_s_pd_sbs_rvi_sccl_2_2/143-1165672-2295435

closed 2024-02-20 by melanielaporte