PubInv/moonrat #34: Feature Request: Fuses for Safety??

magnetictheory, 2023-05-27:
Do we need to have current limit in the product so the fuses do not blow??

RobertLRead, 2023-05-30:
I will set @FORREST ERICKSON ***@***.***> weigh in on
this, but think not.

Moonrat is a low-voltage, modestly low-amperage system. Fuses are usually
not required for that.

However, Lithium Ion batteries can start fires, and physical damage can
create a short circuit. Some sort of break if we draw too many amps is a
good idea; I think Lee knows some tricks beyond installing changable fuses.

ForrestErickson, 2023-05-30:
A deliberate step in the produce design process is FEMA and acronym for Failure Effect Mode Analysis. And for Safety compliance such as UL 60950 .  However when I googled this I find it was a standard approximately for " audio/video, IT, and communication products".

So I asked ChatGPT my first ever real questions first about the history of UL 950 and eventualy got to this helpful (if correct) answer:

> Thanks. You wrote, in part, "...It's important to stay updated with the latest standards and regulations in your specific region or country"
> 
> What would be the relevant safety standard for a device which incubates water samples to test for biological contaminations. Incubation temperature is approximately human body temperature. Power in the heater is approximately 5V at 2.5 A  or 13 watts.
> ChatGPT
> For a device that incubates water samples to test for biological contaminations, the relevant safety standard would depend on the specific type and classification of the device. Generally, such devices would fall under the category of laboratory equipment or medical devices, and the safety standards applicable would vary accordingly.
> 
> In the United States, the safety standards for medical devices are typically governed by the Food and Drug Administration (FDA). The specific standard that may be relevant to your described device is IEC 61010-1, which is the international standard for safety requirements for electrical equipment for measurement, control, and laboratory use. In the United States, this standard is typically adopted as UL 61010-1.
> 
> UL 61010-1/IEC 61010-1 covers a wide range of laboratory equipment and sets requirements for electrical safety, protection against electric shock, temperature limits, mechanical hazards, and other aspects relevant to the safety of electrical equipment used in laboratories and medical settings.
> 
> It's important to consult with regulatory bodies or safety certification organizations like UL or the FDA to ensure compliance with the most up-to-date standards and regulations for your specific device and its intended use. They can provide guidance and specific requirements to ensure the safety of your device.

ForrestErickson, 2023-05-30:
Leaving a question at the web site hosting the EN 61010

http://dfv-technologie.com/clickandbuilds/DFVTechnologie2016/index.php/fr/contact

> I am a volunteer in the United States for a nonprofit designing a laboratory device which will incubate biological samples at near human body temperatures and have been trying to identify safety regulator requirements and EN 61010 may be a candidate.
> 
> I have found what appears to be that document at your hosted web site:
> 
> http://dfv-technologie.com/doc_technique_dfv/normes_afnor/CEM/EN%2061010-1%202010%20Electrical%20safety%20for%20laboratory%20equipment%20-%20General.pdf
> 
> Is my use of this document your web site hosts consistent with it's copywrite?

ForrestErickson, 2023-09-21:
As of 20230920 there is now a fuse in the schematic and on the PCB. IIRC it is a 6A fuse which should not cause nuance trips for the anticipated 2.5A heater load but should protect a batter pack against short circuit loads.

melanielaporte, 2023-11-12:
Complete.

closed 2023-11-12 by melanielaporte