PubInv/moonrat #37: Component Notes for moonratII

ForrestErickson, 2023-06-22:
## Notes as we make the KiCad Schematic for moonratII

The tac switch:
https://www.amazon.com/Momentary-Tactile-Through-Breadboard-Friendly/dp/B07WF76VHT/

Need a foot print. Start with Button_Switch_THT:SW_PUSH_6mm perhaps.

ForrestErickson, 2023-06-22:
The OLED from aliexpress also seams to be the same (Module 12864) as this one at amazon:  
https://www.amazon.com/UCTRONICS-SSD1306-Self-Luminous-Display-Raspberry/dp/B072Q2X2LL

ForrestErickson, 2023-06-22:
While this part is wrong it might be a part with which to start to make a footprint for the OLED display of our choice.
[image]

ForrestErickson, 2023-06-22:
@melanielaporte 

Some possible long lead items to get from Asia:
[image]

[Shop1100385546 Store](https://www.aliexpress.com/store/1100385547)

[10pcs dip 6P 10P 20P 26P 34P 40P PIN 2.54MM pitch MALE SOCKET straight idc box headers PCB CONNECTOR DOUBLE ROW DC3 HEADER](https://www.aliexpress.com/item/3256804080177769.html?mp=1)
$0.96
15-day delivery on $5.00
+ Shipping：$1.97
1

[E-Smartworld Store](https://www.aliexpress.com/store/1101542808)

[1Pcs 2.54mm Pitch JTAG ISP Gray Flat Ribbon Data Cable FC-6/8/10/12/14/16/18/20/24/26/30/34/40/50 Pin FOR DC3 F/F IDC Box Header](https://www.aliexpress.com/item/3256804576275377.html?mp=1)
$2.73
15-day delivery on $5.00
+ Shipping：$1.34
1

[Locheuk Connector Store](https://www.aliexpress.com/store/1101281897)
Get discount

[2.54mm/0.1" Pitch PCB Screw Terminal Block Connector 2P 3P 4P 5P 6P 7P 8P 9P 10P 12P 16Pin Terminals 150V 6A for 26-18AWG Cable](https://www.aliexpress.com/item/2255800681269043.html?mp=1)
$3.75
15-day delivery on $5.00
+ Shipping：$2.91
2

[Win win.](https://www.aliexpress.com/store/1100787794)
Get discount

[0.96 inch IIC Serial 4pin White/Blue/Yellow Blue/Yellow OLED Display Module 128X64 12864 LCD Screen Board for arduino oled](https://www.aliexpress.com/item/2251832770994631.html?mp=1)
$1.47
15-day delivery on $5.00
+ Shipping：$2.32
10
Summary
Subtotal
$25.89
Shipping fee
$8.54
Total
$34.43

ForrestErickson, 2023-07-30:
@RobertLRead ,
Is the design asumption to have the two color OLED display with yellow on top and blue below?
I have a vague recollection of seeing that on the moonrat somewhere but do not remember for certain.
Image:
[image]

Lee
CC: @melanielaporte

ForrestErickson, 2023-07-30:
@melanielaporte ,
Note: 
Specifications for UCTRONICS OLED display:
https://www.uctronics.com/display/uctronics-0-96-inch-oled-module-12864-128x64-yellow-blue-ssd1306-driver-i2c-serial-self-luminous-display-board-for-arduino-raspberry-pi.html

Dimension drawing for UCTRONICS OLED display:
https://www.uctronics.com/download/Mechanical_Drawing/U602602_Dimension_New.jpg

Interface instructions:
https://www.uctronics.com/download/Amazon/U602602.pdf

RobertLRead, 2023-07-30:
No it does use that but that was just what we had on hand. I have no strong
design preference on the OLED as long as it can draw a small temp graph.

ForrestErickson, 2023-07-31:
Regarding Digikey source for Tac switch Lee got an email:

6/27/2023 10:33 AM
> Hi I was able to find this one.
> SW798TB-ND  [B3F-6050 Omron Electronics Inc-EMC Div | Switches | DigiKey](https://www.digikey.com/en/products/detail/omron-electronics-inc-emc-div/B3F-6050/5700024)
> I hope it fits your needs.
> T4576700 is the technical reference for this call.

ForrestErickson, 2023-11-04:
@melanielaporte 

A quick review please and then I think you can close this out with a few notes about how the PCB satisfies the requirements above.

melanielaporte, 2023-11-12:
Completed and discussed.

closed 2023-11-12 by melanielaporte