PubInv/moonrat #47: Make / Find Schematic Symbol and PCB Footprint for Breadboard Friendly Switch

ForrestErickson, 2023-07-28:
For this part:
https://www.amazon.com/Momentary-Tactile-Through-Breadboard-Friendly/dp/B07WF76VHT/

[image]

ForrestErickson, 2023-07-30:
Message to JLCPCB,

> I want to design a PCB using this switch like what I buy for prototyping from Amazon> 
> https://www.amazon.com/Momentary-Tactile-Through-Breadboard-Friendly/dp/B07WF76VHT/
> I am not able to find a JLCPCB equivalent but hope you can find one.
> Send answer to forresterickson@comcast.net

ForrestErickson, 2023-07-31:
I had difficulty with the new part request and the tech support helped me as follows:

> Michael 05:06 AM
> Welcome back, may I help you?
> 2999181A 05:08 AM
> I am trying to use the Parts Manager to request a New Part
> Name(Account #):
> 2999181A
> What do you need help with?
> Others
> avatar
> Michael 05:08 AM
> hi
> 2999181A 05:09 AM
> I am trying to use the Parts Manager New Parts Request.
> I do not understand the user interface on the JLCPCB web site.
> There is a button to "Submit a Request. However there is no form into which I can submit information in a way that makes sense.
> avatar
> Michael 05:10 AM
> please fill in keyword or JLCPCB part number to search it first.
> 2999181A 05:11 AM
> I was told in an email from JLCPCB that this is the form to reqeust a NEW part number. There is no JLCPCB part number that I can enter.
> Message: I want to design a PCB using this switch like what I buy for prototyping from Amazon
> 
> https://www.amazon.com/Momentary-Tactile-Through-Breadboard-Friendly/dp/B07WF76VHT/
> 
> I am not able to find a JLCPCB equivalent but hope you can find one.
> 
> Send answer to [forresterickson@comcast.net](mailto:forresterickson@comcast.net)
> The last post in this thread is exactly the message I sent to JLCPCB
> 
> This is the answer I got:
> Dear Sir,
> 
> 
> 
> Please submit a new part request as the below picture, then our Purchasing Department will review and reply to you soon.
> 
> 
> 
> ﻿
> 
> 
> 
> Have a nice day ^_^
> 
> 
> 
> Eric
> 
> 
> 
> https://d33v4339jhl8k0.cloudfront.net/users/516734.260580.png
> Eric He
> Web: [JLCPCB.COM](http://jlcpcb.com/)
> Email: [eric@jlcpcb.com](mailto:eric@jlcpcb.com)
> I am not able to "Please submit a new part request..."
> image.png
> avatar
> Michael 05:15 AM
> image.png
> you could search with B07WF76VHT
> and then you will be able to fill in the info
> image.png
> 2999181A 05:21 AM
> OK I think I see it now. I am starting. Please stay with me.
> avatar
> Michael 05:21 AM
> alright
> 2999181A 05:23 AM
> image.png
> I have submitted the form. for:
> 
> Manufacturer
> Omron Electronics Inc-EMC Div
> Manufacturer Product Number
> B3F-6050
> Description
> SWITCH TACTILE SPST-NO 0.05A 24V
> I assume it is at the end of the business day there at JLCPCB and it will not be until tomorrow before your purchasing department can start on this. Am I correct?
> Read
> avatar
> Michael 05:28 AM
> yes. you could check the status of it tomorrow.
> 2999181A 05:29 AM
> Thanks for helping me with the form.
> 
> I hope you can pass on to people who design the web page the difficulty I had so that they might improve the form.
> 
> You did your job well.
> Xièxiè

ForrestErickson, 2023-07-31:
Information for making a foot print for the Omron B3F-6050
[image]

ForrestErickson, 2023-09-09:
@melanielaporte 
Any chance you can work on this soon?

ForrestErickson, 2023-10-15:
Time to give up.
Use the four pin part:
https://jlcpcb.com/partdetail/OmronElectronics-B3F1052/C87037

melanielaporte, 2023-11-12:
Will not update.

closed 2023-11-12 by melanielaporte