PubInv/moonrat #48: What Batter / Power Supply To Use?  Power Tool?  Wall Supply?

ForrestErickson, 2023-07-28:
Power tool batteries may be common world wide?
[image]

Wall supplies too?

ForrestErickson, 2023-07-28:
@RobertLRead ,
What do you know or have sources regarding the availability of power tool batteries in low income locations?

The Dewalt batter and adapter converter above would be a way to use such batteries that may be common where ever building happens.

Please advise,
Lee
cc: @melanielaporte

RobertLRead, 2023-07-28:
I think this is an interesting idea. I assume these are at least as available as the 12V USB power banks which we were planning to use, and probably recharge faster. 

However, we have to be careful if power tool batteries are not standardized wrt voltage. If we go above 12V, we probably need to carefully manage the duty cycle of the power, because we want the heating to remain gentle---you don't want any thing very hot inside the incubation chamber, even if eventually the heat will even out, because a hot spot can sterilize the sample.

Also, USB-based charging is equally ubiquitous. So I'm unless power batteries are highly standardized, I'm not sure there is much advantage here. For example, when I was in Tanzania in the larger towns (Kigoma) you could by a full range of hardware, and I'm sure you could get drills. BUT I don't know that any one brand is completely ubiquitous across different nations; and if multiple brands are not interchangeable, that doesn't help us.

@ForrestErickson @melanielaporte

melanielaporte, 2024-02-20:
@ForrestErickson @RobertLRead Does this topic need another discussion for PCB v2 design and order?

melanielaporte, 2024-09-18:
@ForrestErickson @RobertLRead Does this topic need another discussion for the next design generation design?

ForrestErickson, 2024-09-19:
Regarding, "Does this topic need another discussion for the next design generation design?"
I think not. At lease until we find that the batteries used by the Guadalajara team are hard to come by.

I suggest Melanie close this as will not fix.

melanielaporte, 2024-12-06:
Will not do.

closed 2024-12-06 by melanielaporte