PubInv/moonrat #49: OLED Displays Modules ROHS from JLCPCB?

ForrestErickson, 2023-07-28:
See the part:
https://jlcpcb.com/partdetail/Hs-HS96L03W2C03/C5248080

Date sheet: https://datasheet.lcsc.com/lcsc/2305050946_HS-HS96L03W2C03_C5248080.pdf

ForrestErickson, 2023-07-28:
**Lee had an online chat with JLCPCB regarding how to assemble this.**

Hello Anna. For JLCPCB Part # C5248080 can you confirm there is a display on the other size of the PCB?
The image shows only what I would think is the back of the board. The side with out the display.
See: https://jlcpcb.com/partdetail/Hs-HS96L03W2C03/C5248080
avatar
Anna 10:35 PM
a moment please, let me have a look
2999181A 10:35 PM
I understand.
avatar
Anna 10:37 PM
image.png
2999181A 10:37 PM
That is great. Thanks.
Can you tell me if and how JLCPCB can mount this to a PCB which I design?
I assume I need four screws, four insulating spacers and four nuts and a four pin header to make the signal connections.
Is such assembly something JLCPCB can do?
I am going to design a PCB in KiCAD with which I am fluent. Just not sure if I do so that JLCPCB can assemble it.
avatar
Anna 10:39 PM
yes, please kindly include the insulating spacers, nuts and four pin header in the BOM list and CPL file as well
2999181A 10:40 PM
OK I will have to check the JLCPCB parts for something compatible.

Is it possible that JLCPCB can identify screws, a spacer about 1/8" or about 3mm thick and nut?
avatar
Anna 10:41 PM
https://jlcpcb.com/parts
please kindly search for them in our library ~
2999181A 10:42 PM
The drawing for the part shows a 2mm mounting hole and so I assume I need a slightly smaller screw. Living in the USA I am not famiaiure with such small metric hardware and do not know common terms with which to search.
I have tried searching for a 1.5mm screw with no success.
avatar
Anna 10:44 PM
would you kindly try if there are 1.5mm screw in other parts supplier? if you can find the part from other global sourcing supplier , you can try to use our global sourcing service to pre order the part ,detail step link: https://jlcpcb.com/help/article/60-How-to-use-JLCPCB-Global-Sourcing-Parts-Service
2999181A 10:44 PM
I do not find any screw.
I will read this and try to figure it out.

Would it be possible to have JLCPCB build the boards and put all of the other parts on and ship with the OLED diplay not installed and I will install it? The OLD Display would need to be in some bubble wrap for example.
We will likely be ordering 15 or 20 assemblies (PCBs with parts) and so would want 15 or 20 OLED diplays too. I would get the screw, spacer and nut here in the USA.
avatar
Anna 10:48 PM
I am afraid not, we cannot support sell components separately, on the other hand, if we ship the parts together with PCB, they will probably hold by the customs
2999181A 10:50 PM
Ok I think I understand the potential problems. Thank you for helping me understand this.

Displays are available here easy but I was trying to figure out how I have have JLCPCB complete the full build which would be a nice goal for all of us.
avatar
Anna 10:51 PM
thanks for you understanding
2999181A 10:52 PM
Thank you and have a good day. I assume it is Friday and late morning there.

I am up late on my Thursday and it is so nice to have fast response from half way around the world.
avatar
Anna 10:52 PM
yes, it is Friday and late morning here ~
have a lovely evening ~
2999181A 10:53 PM
Xièxiè
Read
avatar
Anna 10:53 PM
😄
2999181A 10:53 PM
Bye.
Delivered

ForrestErickson, 2023-07-28:
URL to the

ForrestErickson, 2023-07-30:
Lee decide to ask as a question: 

I want to put C5248080 on a PCB.   But a 4 position header and four mounting screws, nuts and spacers are probably needed.  Can you advise what I need in my PCB and my BOM to use the OLED display: C5248080?

Send information to my email at forresterickson@comcast.net please.

RobertLRead, 2023-07-30:
> Send information to my email at ***@***.*** please.

Are you asking me?

ForrestErickson, 2023-07-30:
@RobertLRead regarding, "Are you asking me?"  No I was documenting a question to JLCPCB. My bad for not explaning. I was kind of punchy by then.

ForrestErickson, 2023-10-15:
@melanielaporte 
Got your email of "10/14/2023 7:33 PM" with the question "Would you check my OLED footprint tomorrow? I'll then add it to the schematic. "

When I look at the repository log in updated clone I do not see any check in related tot eh OLED footprint so I assume that at 4:57 Am this morning I do not have the footprint to review. 

Check in the changes when you can.

I will be tied up from 10:30 to 12:30 today, Sunday the 15 but can work before or after if you call me.

Lee

melanielaporte, 2023-11-12:
Melanie designed a footprint for JCLPCB module.

closed 2023-11-12 by melanielaporte