PubInv/moonrat #55: When ordering the PCB, ask manufacturer to V cut or score/gouge the PCB so we can snap it manually = 2 boards instead 1

melanielaporte, 2023-08-01:


ForrestErickson, 2023-11-06:
The Controller side must be seperated from the heater side. A V-Cut is the easy way to do this.

[image]
The white line on Silk Screen currently shows the V-Cut but is just a place holder till we identify a good way to work with JLCPCB to identify , specify the V-cut.

Lee asked JLCPCB for advise and was referred to this link

https://jlcpcb.com/help/article/96-PCB-Panelization

melanielaporte, 2023-11-12:
Vcut has been added to the PCB.

closed 2023-11-12 by melanielaporte

melanielaporte, 2024-03-07:
Exact v-cut/v-score wording: 1. Please confirm V-cut is feasible on User 1 layer 
2. MF12 (1x4 header) should be soldered to J2 (OLED display) 
3. MF 4-6, 8 (round nuts) mounted to OLED display 4. MF1-3 (keycaps) to go on SW1-3