PubInv/moonrat #73: apple deletes kicad, chrome, and git files each time I restart - Melanie

melanielaporte, 2023-09-07:


RobertLRead, 2023-09-07:
Melanie, do you want to do a screenshare to try to work through this?

melanielaporte, 2023-09-07:
I would love to! Lee and I are supposed to meet tonight at 7pm but maybe this takes precedence.

ForrestErickson, 2023-09-08:
What is the status now?

RobertLRead, 2023-09-08:
We got past it --- I think. We should wait for Melanie to close it.

closed 2023-09-08 by melanielaporte

ForrestErickson, 2023-09-08:
@RobertLRead 
Regarding, "We got past it -"  What was "it"?

File & folder permissions I am guessing but don't make me guess.
Is it the permissions in the folder in which work was being saved or was it permissions on folders where KiCad or other reside?

Same for GIT.

RobertLRead, 2023-09-08:
I think the installers were being put in the "dock" instead of the app,
which made it easy to constantly reinstall things.

Additionally, on installation, the files may not have been dragged into the
"application" folder.  Also, the security question asked by Apple when an
application is opened for the first time was not answered.