PubInv/moonrat #87: Buzzer Hole Spacing or 3D Model Wrong

ForrestErickson, 2023-10-15:
The buzzer hole spacing appears to be wrong on the 3D view.

[image]

Does the PCB match the part?
Is the 3D render correct to the part?

melanielaporte, 2023-10-15:
Check updated buzzer PCB position.

melanielaporte, 2023-10-21:
10/21 We reviewed and confirmed that the error is ONLY the 3D model and the PCB spacing is 7.6mm. 
GPAD was built with this buzzer - https://www.digikey.com/en/products/detail/soberton-inc/WST-1210T/16354828?s=N4IgTCBcDaICwGYEFoDqBlAKsgjGHADNgHIAiIAugL5A
We are going forward with the GPAD. 
I will order 5 assembles and 5 raw boards, and 10 buzzer parts as back ups and for experiments.

melanielaporte, 2023-10-21:
To do: Lee will make a new 3D model TBD.

ForrestErickson, 2023-11-04:
Found a model here: https://grabcad.com/library/buzzer-12mm-continuo-1

[image]

Loaded into PublicInvention Library. 

@melanielaporte  will will need to connect the MoonratII with the Public Invention KiCAD library at some point. It is not cirtical for the order of PCBs however.

melanielaporte, 2023-11-12:
Changed buzzer manufacturer to 
- Distributor 1 JCLPCB C360615
https://jlcpcb.com/partdetail/ss-SUN_12095_5VPA76/C360615#:~:text=Datasheet-,Download,-Source
and 
- Distributor 2 Adafruit P1536A

ForrestErickson, 2023-12-01:
@melanielaporte 
I do not remember fixing this and suspect you did.
The buzzer 3D model is now correct.
[image]

Did you fix it?

Regardless you may close this issue.

closed 2024-02-20 by melanielaporte