PubInv/moonrat #97: Clearing Schematic Errors

ForrestErickson, 2023-11-06:
Of the 33 schematic errors 30 are the pins on these connectors 

[image]

These pins need to be marked with the "x" no connect.

ForrestErickson, 2023-11-06:
Set no connect marker on these pins and rerun ERC.

[image]

Only three Errors left!

ForrestErickson, 2023-11-06:
For VCCQ noi driven by a power pin.

[image]

Try to clear by adding a power flag.
[image]

Rerun ERC and it worked.
[image]

ForrestErickson, 2023-11-06:
For the error
[power_pin_not_driven]: Input Power pin not driven by any Output Power pins
    ; Severity: error
    @(9600.00 mils, 5600.00 mils): Symbol U1 Pin 4 [GND, Power input, Line]

[image]

Try adding power flag.  Worked cleared the U1 Pin 4 error.
[image]

ForrestErickson, 2023-11-06:
Lets add a power flag to +5P

[image]

That cleared the last error.

melanielaporte, 2023-11-12:
As intended, updated by marking pins not connected.

closed 2023-11-12 by melanielaporte