PubInv/moonrat #113: Find another TMP for the PCB with more resolution.

melanielaporte, 2023-11-14:


ForrestErickson, 2023-11-14:
@RobertLRead 

Can you reference the temperature tolerance requirement for the incubator materials of which you spoke during the meeting?
FYI, The TMP36 data sheet says +/- 2degreeC accuracy
[image]

CC: @melanielaporte

RobertLRead, 2023-11-14:
These instructions specify 35C  +-1 C:

https://www.neogen.com/globalassets/pim/assets/original/10044/34871738398_int.pdf

Here is the 3M E. coil interpretation guide:

https://media.neogen.com/m/3ddc380f58cfa4ed

HJGV05, 2023-11-17:
I have done some research on the best sensors for the application and here are my conclusions:
- It is important and necessary to change the TMP family sensor (TMP35-37) since its accuracy is ± 2°C above room temperature (see attached data sheet), which is not feasible since Petrifilm's datasheet (https://www.neogen.com/globalassets/pim/assets/original/10044/34871738398_int.pdf) mentions that the sample incubation should be 24 hours ± 2 hours at 35°C ± 1°C.
- The best commercial analog sensor that meets the specifications are the sensors from the LM family, specifically the LM35 (data sheet attached). This sensor is a precision temperature sensor whose accuracy is ± 0.5°C at 25°C and ± 0.25°C at room temperature. In addition, the operation current is less than 60-μA, so the maximum self-heating of the sensor is less than 0.08°C,  which translates to less sensor heating noise in the output signal.
- The LM35 sensor does not require prior calibration, the library and documentation on Arduino is very extensive and easy to find (https://www.arduino.cc/reference/en/libraries/lm35-sensor/ ) and does not interfere with the current PCB schematic. However, its only drawback is that it requires a higher operating voltage than the TMP family, the recommended supply voltage is 4 V to 5.5 V, but it can be powered from 4 V to 30 V. While this is not a drawback for the current schematic, it is an important note in case that in the future it is desired to work with low-energy devices (powered with 3.3 V).
- For the previous point, it is recommended in the future to change the analog port for an I2C port (or include it as an additional one) so that integrated sensors for temperature, humidity and pressure can be used, such as the BME280 (data sheet attached), which has an I2C communication protocol and a supply voltage of 1.7 V to 3.6 V, and whose library for integration into the Arduino system is available through the open source company Adafruit. For more information on temperature sensor comparisons, please see https://randomnerdtutorials.com/dht11-vs-dht22-vs-lm35-vs-ds18b20-vs-bme280-vs-bmp180/ 
- As a last point, I am concerned about 3M's requirements for the preservation of petrifilm samples before use, since they must be stored at a temperature less than or equal to 8°C, has there been any discussion about how this requirement will be met?

[bme280.pdf](https://github.com/PubInv/moonrat/files/13392409/bme280.pdf)
[lm35.pdf](https://github.com/PubInv/moonrat/files/13392411/lm35.pdf)
[TMP35.PDF](https://github.com/PubInv/moonrat/files/13392412/TMP35.PDF)

RobertLRead, 2023-11-17:
This is fantastic research, Horacio! I think adding an I2C port would be
nice, but it might change the wire harness, and I'm not sure we want to do
that. We might be able to survive with the LM35 for this version.

I keep my Petrifilms in the fridge. We can inform researchers to do the
same, but generally speaking I don't believe there is a problem with
leaving them at room temperature for a few days (that is, close to a human
being, which pretty much guarantees they won't get hotter than a human can
tolerate.) I don't believe it is a requirement of this project to address
that problem.

Someday someone may build a chamber that can both heat AND cool. That would
be a valuable scientific instrument but is about 20 times harder than what
we are trying to do.

ForrestErickson, 2023-11-17:
Just thinking out loud.

### Arduino A2D
The Arduino UNO has an A2D resolution of 1023 steps. If the reverence voltage is the nominal 5V this is 5/1023 step or 4.89mV.

### TMP36 and TMP37
The TMP36 specification says 10 mV/C for the conversion. This implies our conversion step size is 4.89mV/(10mV/C) = 0.489 degree C.
The specification says -.5 degree linearity over the temperature range.
Could we simply calibrate each measurement device in an ice water bath and store the calibration in EEPROM?

The TMP37 has twice the gain so the temperature step size would be 0.244 degree C.
https://www.analog.com/media/en/technical-documentation/data-sheets/tmp35_36_37.pdf

### Arduino A2D Reference
Like many embedded A2Ds the resolution can be improved by matching the conversion range to the expected sensor analog voltage. From the TMP36 TMP37 data sheet we can see that even for temperatures as high as 50C the sensor output voltage is less than 1V.
[image]

If we bias the Artuino UNO reference input pin to for example 1/2 Vcc we gain an additional factor two temperature resolution so we might have 0.122 degree C step size.

### PCB Implications
@melanielaporte  This would require we add two resistors to divide the supply voltage in half and connect the result to AREF input on the UNO.  A 0.1 capacitor there would be well advised too.

### Supply at JLCPCB. 
Unfortunately there is no supply of the TMP37 in the SOT23-5 package at JLCPCB.
We can get them from Digikey at $2.40000 single piece quantity in stock.

RobertLRead, 2023-11-17:
This could be tried---we would have to attempt it and test it.

But why is this better than shifting to an LM35?

ForrestErickson, 2023-11-17:
Regarding, "But why is this better than shifting to an LM35?"
Caution: I have no direct experience with any of these parts.  Everything I am writing is based on reading the specifications and of course the first time through I may read what I want to read not what is there.

The TMP37 has twice the gain (20mV/C) of the LM35 which is a big help for resolution. Not for accuracy.  We would have to do accuracy calibration in firmware at assembly time.

The LM35 is specified (?much?) tighter accuracy. 
There is no SOT32 package for the TM35 if we go with the LM35 we have to change the footprint to SOIC and add the series resistor per this diagram.|
[image]
 

The LM35  is available from JLCPLC.
LM35DMX/NOPB
Manufacturer : Texas Instruments
MFR.Part # : LM35DMX/NOPB
JLCPCB Part #: C2683367
Package: SOP-8

HJGV05, 2023-11-18:
> This is fantastic research, Horacio! I think adding an I2C port would be nice, but it might change the wire harness, and I'm not sure we want to do that. We might be able to survive with the LM35 for this version. I keep my Petrifilms in the fridge. We can inform researchers to do the same, but generally speaking I don't believe there is a problem with leaving them at room temperature for a few days (that is, close to a human being, which pretty much guarantees they won't get hotter than a human can tolerate.) I don't believe it is a requirement of this project to address that problem. Someday someone may build a chamber that can both heat AND cool. That would be a valuable scientific instrument but is about 20 times harder than what we are trying to do.
> […](#)
> -- Robert L. Read, PhD Twitter: @RobertLeeRead @pubinvention Public Invention: https://www.pubinv.org Join Our Mailing list: ***@***.*** YouTube: https://www.youtube.com/channel/UCJQg_dkDY3KTP1ybugYwReg Medium: ***@***.***

Thank you very much Robert, I'm glad you liked it. Sure, regarding the current version, I think it's enough with the analog port without changing any PCB configuration. Regarding future versions, since only two additional ports are added to connect to the shield that goes to the Arduino (ports SDA and SCL) is sufficient to replace it with a 12-pin harness (see attached image) which is very common and cheap because of their usage in auto stereos. The advantage of this change is that the I2C protocol allows the connection of multiple sensors through the same bus since each of them has an identifier, so the PCB of the sensor board could include many more ports. Of course, this is only an idea for the future.
[image]

HJGV05, 2023-11-18:
> Just thinking out loud.
>
> ### Arduino A2D
> The Arduino UNO has an A2D resolution of 1023 steps. If the reverence voltage is the nominal 5V this is 5/1023 step or 4.89mV.
>
> ### TMP36 and TMP37
> The TMP36 [...]

Thank you very much for your clarification Mr. Lee, it is a very interesting observation. I have doubts about the reliability of calibrating the TMP family sensors using ice water and about the complications for the target community when performing this process. While it is true that we might lose reading resolution, I think accuracy is more important for the task at hand. About the inclusion of the additional RL circuit for higher capacitive loads, I think it is not necessary for the temperature range that will be handled, a better analysis would have to be done about it and even some tests on a protoboard to see how it behaves in the lab.

HJGV05, 2023-11-18:
Finally, regarding petrifilm samples, the 3M company officially mentions that the plates can be up to 72 hours at room temperature (35 °C, see https://multimedia.3m.com/mws/media/243532O/tb-001-01-shipping-temperature-of-3m-petrifilm-plates.pdf) without compromising the integrity and reliability of the plate, so I feel more confident about this requirement. I finally include the official source link on how to take the samples, and the times and temperatures required for different types of samples, so we can even think in the future about an interface that allows us to select the type of sample to incubate and that the incubator can maintain this reference according to the selected sample, greetings! https://multimedia.3m.com/mws/media/777561O/3m-petrifilm-plates-sample-plates.pdf

ForrestErickson, 2023-11-18:
@HJGV05 
Regarding, "12-pin harness (see attached image) which is very common and cheap because of their usage in auto stereos."
**Nice find.**  I hope to get out to my local auto parts store and become more up to date.  It is about 1978 since I last purchased an auto stereo.

Can you identify vendor information (Name and part number and URL) on both the male and female for this connector with wires?  The trailer light parts I had identified can be found on Amazon however I never noticed reference to any standard. 

The root reason I added to our schematic and PCB the ten position ribbon cable is that I knew these could be in principle be constructed out of materials from distributors such as Mouser.com and Digikey.com and would always be available through such vendors.  As a method of repeated connect and disconnect the ribbon cable is not convenient and would impact design of one or both the enclosures of the controller or the incubator heater since there is no inline decoupling of the cable. 

In the schematic at present are: 
* For the 10 conductor IDC cable the following amazon part number:
https://www.amazon.com/gp/product/B01DP13EDA/ref=ox_sc_saved_title_2?smid=A64W1E1ZZHST0&th=1

* For the 6 Pin Socket and Plug
 https://www.amazon.com/807-Connectors-Extension-Connector-Way-Socket/dp/B07SR28CG4/ref=sr_1_1?th=1
 
And I see there exists 8 pin connectors too.
https://www.amazon.com/gp/product/B07RNN85TX/ref=ox_sc_act_title_1?smid=A3W23XOXVE5ZM8&th=1
This cable would accommodate I2C signals.

CC: 
@melanielaporte

HJGV05, 2023-11-18:
@RobertLRead 
Sure, with pleasure. This type of 12-pin harness is usually used especially by the Pioneer brand (e.g. https://www.ebay.com/itm/190365766282?hash=item2c52af128a:g:Z0MAAOSwUKxYZVWo&amdata=enc%3AAQAIAAAA8FsleXMxXFF9u8YOuIlQuY69HqN0yzR2AexpdikLdo2%2FAaZMFdFFIy8VdBzWDnRBKDqGK1vUFtcjEGj7mazokdp8K6rocoDHEGb3ANWjKXwf9gC%2BfV9nORGD1meUuAC9Z0VDhM8qqyuJ%2BA7Tfh60506uXJ%2BeXGDFOXml7w18fR4rEkLHOVubAFtWTDu6JMr8JkFu32Do%2FYKQACamzqLA46X%2B49TcKHbyBruXfqQbqVCgH8P5P5267XQ3E3LRmzGsKqADKHMBcIQMbHnzALx3veemK47CYAXk%2B56%2B2RSArUMS499651sK0xY%2B1EDc6w3dhg%3D%3D%7Ctkp%3ABk9SR6C0nsv8Yg). However, I think it would even be a better option to use cables with IDC connectors like the ones mentioned by Mr. Lee, but in their 12 or more pins version. The reason for this is that they are affordable compared to the cables available on Amazon, for example, and they are accessible to the global market. For reference, Aliexpress, which ships to several places in the world, has configurations of both male and female-female harness connectors configurable from 6 to 22 pins and from 10cm to 50 cm at a very affordable cost. Specifically speaking, a 10-pack of 2X6-pin male IDC connectors (12 pins in total) costs 1.68 USD (plus shipping which can be free depending on the quantity, see for example https://es.aliexpress.com/item/1005002397737993.html?spm=a2g0o.best.0.0.172122aeMcgCAQ&mp=1&gatewayAdapt=glo2esp ) while a 2x6-pin 30cm harness costs 1.75 USD plus shipping (same situation, see https://es.aliexpress.com/item/1005006154430582.html?spm=a2g0o.best.0.0.172122aeMcgCAQ&mp=1&gatewayAdapt=glo2esp). I think this configuration and supplier is very useful because it provides a distribution channel available in several parts of the world, a situation that may be different through Amazon USA.

RobertLRead, 2023-11-19:
This is good research --- but I will allow you and Lee and Melanie to make
the final decision on this.

ForrestErickson, 2023-11-19:
Regarding the choice of cables.
Our choice will impact what we do with the incubator. 

[image]

If the incubator is an off the shelf soup thermos as Robert has suggested from time to time we have to penetrate the lid or the body of the thermos to pass the cable through.  There are consequences some what different to each decision. 
If we degrade the insulation factor by penetrating the main body we have degraded the whole system to a greater extent.

If we penetrate only the lid there may be less degradation of insulation but the cables will be subject to handling and rotation every time the lid is put on and off the main body.

The compromises or trad off will be how the cable goes through what ever penetration we make. How it is strain relived. How it is sealed. 

I have only vague ideas.

Inside the incubator something must hold the sample and the heater and the temperature sensor.

Again, I have only vague ideas.

There is also a battery and battery holder to figure in to all this.

Other:
The molded trailer cables connect between the controller enclosure and the incubator. The molded trailer cables can resist water. 
The Audio cable connectors are open to the back. and would not resist water if used as a method of connecting/ disconnecting.

CC: @melanielaporte

RobertLRead, 2023-11-19:
The problem of the lid twisting is solvable by twisting the body, not the
lid.

ForrestErickson, 2023-11-19:
Regarding, "The problem of the lid twisting is solvable by twisting the body, not the lid."

Which has implications for what ever holds the heater and the sample with media and the temperature sensor.  I think they must also then attach to the lid.  

A total incubator mechanical proposal is necessary.

As I said, Our choice will impact what we do with the incubator.

I have only vague ideas.  I do not have a good mental image of how the device that visited Tanzania was realized.

HJGV05, 2023-11-20:
> Regarding the choice of cables. Our choice will impact what we do with the incubator.
>
> [image]
>
> If the incubator is an off the shelf soup thermos as Robert has suggested from time to time we have to [...]

Indeed, I think this is an important mechanical and design problem to take into account, I had mentioned when I knew the project that it would also be a good option to include a Bluetooth low energy module (BLE) instead of making a physical interface that was mounted on the incubator, this way the sealing of the incubator is not compromised and we can make a mobile app that has the role of both monitor and digital interface with a low energy cost. Even a buzzer could be included in the incubator circuit that emits a sound periodically to indicate that everything is in order (or not) and thus not have to be monitoring the cell phone at all times. We could open a new issue to discuss this topic.

RobertLRead, 2023-11-20:
I don't quite understand this --- you have to get the power wires into the
heating cloth somehow, unless you put the battery in the incubator, which
seems like a bad idea. So you are going to have to drill a hole into the
Thermos bottle one way or another.

My personal approach, at least for Petrifilms, is to drill a whole in the
cap, and put the wires through that, and then seal the the hole back up
with the wires in it.  This is how it works in the system we took to
Tanzania, and it works well.

However, one of the reasons to have the "N-wire interface" (no matter what
N is) is to allow incubation changers of different size. However, being
able to handle up to 40 petrifilms (in stacks of up to 20) in the incubator
is the most important use case.

ForrestErickson, 2023-12-01:
@melanielaporte 
@HJGV05 
@SilviaKarbid 
@kurama79 
@RobertLRead 

With the Rev 1.0 printed wiring assemblies we will have the JLCPCB part number: C129489 which is Analog Devices TMP36GRTZ-REEL7

An system engineering activity and design evaluation we can and should carry out is to characterized ALL of the assemblies we get for accuracy.
I would envision measuring accuracy at 0C with an ice water bath is trivially easy. Harder is a boiling water bath but possible I assume. 
More difficult yet would be to characterize linearity between these two points and especially near body temperature and near the incubation temperature.  This will require some device well calibrated near these temperatures.  
With this data we might develop a calibration methodology (Linear approximation, or look up table or higher order fit) and understand the limitation of our system design.

Until we have the expertise to know how well the TMP36GRTZ-REEL7 design works we are speculating only.

closed 2024-02-20 by melanielaporte