PubInv/moonrat #117: Make or Find 3D model for Switch Key Cap,

ForrestErickson, 2023-11-14:
On 20231114 Lee requested files from Omeron contact form at: https://components.omron.com/us-en/contact-us/inquiry?inquiry_summary=CAD%20data

### The Request
Would like 3D files for B32-1600
Our application is with switch B3F-1052
We want to import 3D file into KiCad which requires STEP and WRL. See: https://klc.kicad.org/model/m2/m2.1/

We also want to import into FreeCAD and so other file formats may be of use.

### Add 3D model to the MoonratII PCB.
Fit to SW1, SE2, and SW3
JLCPCB screen shot of this part.
[image]

melanielaporte, 2023-11-15:
We have been instructed by JCLPCB to place special assembly instructions in the Other Information field to include the keycaps in the board order. Due to the lack of keycap footprint. 
One option is to hand assemble the keycaps instead of JCLPCB.

melanielaporte, 2023-11-15:
@melanielaporte Send JCLPCB 3D renders to better illustrate our goal. 20231116!

ForrestErickson, 2023-12-04:
Notes from drawing on the Key Cap B32-1600
[image]

closed 2023-12-25 by melanielaporte