PubInv/moonrat #146: Firmware, Add and Report I2C Address Scan at Boot

ForrestErickson, 2023-12-04:
The firmware should help identify hardware faults where possible for both manufacturing users and end users.

The firmware should scan the I2C bus for all devices and report those found on the serial UART (USB) port.
The Arduino UNO example libraries has an I2C scan sketch from which to star code.

The firmware of the GPAD has such or similar feature. 
The firmware of the GPAD factory test has some additional features for consideration

ForrestErickson, 2023-12-16:
The OLED is an I2C device.

HJGV05, 2023-12-26:
Good day, it is a good idea to perform a diagnostic and component fault test script. However, I think it would be better to create a separate firmware that the user can run on the board, we can make a diagnostic script that reads and displays all the I2C components connected, check the temperature sensor reading, allow the use the buttons, send a small PWM output signal to test the heater setup and emit some test sound for the buzzer so that it is simple and light to implement without adding more complexity to the main program interface for the incubator, do you like this idea?

ForrestErickson, 2023-12-26:
@HJGV05 

Regarding, ", I think it would be better to create a separate firmware that the user can run on the board,"
Yes a separate "**Factory Test**" firmware is likely to be of value because the user experience can be optimized for the factory test process. 

There is a possibility that a **Trouble Shooting** test firmware is also needed for some kinds of problems. I have from time to time coded up special firmware focused on helping to diagnose problems. Such firmware is often trying to make seeing a problem with an oscilloscope possible. 

I have just added some folders into which such work can be placed.

closed 2024-02-20 by melanielaporte