PubInv/moonrat #170: Write and Submit MoonRat II HardwareX Article

ForrestErickson, 2023-12-28:
The HardwareX article template is in the folder at:
https://github.com/PubInv/moonrat/tree/main/moonratII/HardwareX_moonratII

I hope it will be useful and I am extracting from the template _some_ of the tasks which compose the writing of this article into this check list. These are the mostly technical (not administrative) items.
_The numbering reflects the template which unfortunately does not make a useful table of contents_

- [ ] 1. Hardware in context
- [ ] 2. Hardware description. 
- [ ] 1. Design files 
- [ ] Design Files Summary
- [ ] 2. Bill of Materials
- [ ] 3. Build Instructions
- [ ] 1. Operation Instructions
- [ ] 1. Validation and Characterization
- [ ]

melanielaporte, 2024-09-18:
@melanielaporte

melanielaporte, 2025-01-15:
@RobertLRead @ForrestErickson Please update your progress in this issue.