PubInv/moonrat #178: HardwareX Article,  1. Validation and Characterization

ForrestErickson, 2023-12-28:
From the template,

 

> 1. Validation and Characterization
>  Demonstrate the operation of the hardware and characterize its performance over relevant critical metrics:
> * Demonstrate the use of the hardware for a relevant use case.
> * If possible, characterize performance of the hardware over operational parameters.
> * Create a bulleted list that describes the capabilities (and limitations) of the hardware. For example consider descriptions of load, operation time, spin speed, coefficient of variation, accuracy, precision and etc.

melanielaporte, 2024-09-18:
@RobertLRead

RobertLRead, 2024-09-25:
@SilviaKarbid and team agrees to perform heat resistivity/conductivity analysis of Bento Box. (Rob believes this is very similar to determing wattage needed to hold a temperature difference above ambient.)