PubInv/moonrat #187: Important datasheets and remarks

SilviaKarbid, 2024-01-10:
Heating pad adafruit datasheet
[Ultra Heating Fabric.pdf](https://github.com/PubInv/moonrat/files/13891211/Ultra.Heating.Fabric.pdf)

kurama79, 2024-01-10:
Here is the wiki link of the display [https://wiki.seeedstudio.com/Grove-OLED-Display-0.66-SSD1306_v1.0/](url)

kurama79, 2024-01-10:
And the datasheets of the sensors:
[LM35.PDF](https://github.com/PubInv/moonrat/files/1389
[TMP35.PDF](https://github.com/PubInv/moonrat/files/13891231/TMP35.PDF)
1230/LM35.PDF)

kurama79, 2024-01-10:
And the datasheets of the sensors:
[LM35.PDF](https://github.com/PubInv/m
[TMP35.PDF](https://github.com/PubInv/moonrat/files/13891240/TMP35.PDF)
oonrat/files/13891238/LM35.PDF)

kurama79, 2024-01-10:
And the datasheets of the sensors: 
[LM35.PDF](https://github.com/PubInv/moonrat/files/13891246/LM35.PDF)

kurama79, 2024-01-10:
[TMP35.PDF](https://github.com/PubInv/moonrat/files/13891249/TMP35.PDF)

ForrestErickson, 2024-01-10:
From the KiCad schematic there is the properties view of the OLED display
[image]

JLCPCB part number is: C5248080
If you visit the their web site you can get datasheets for the parts they are providing. In this case this URL:
https://jlcpcb.com/partdetail/Hs-HS96L03W2C03/C5248080

If you look into the repository in this folder under the PCB you will find many of the data sheets for the parts we use. 
It is not formally organized and you have to open and look at files where the file name is not self explanatory
https://github.com/PubInv/moonrat/tree/main/moonratII/pcb_moonratII/parts
The display data sheet is there: [HS96L03W2C03.pdf](https://github.com/PubInv/moonrat/blob/main/moonratII/pcb_moonratII/parts/HS96L03W2C03.pdf)

Hope this helps.

ForrestErickson, 2024-01-10:
@HJGV05 
@SilviaKarbid 
@kurama79 
There is a README.md at the PCB folder here:
https://github.com/PubInv/moonrat/tree/main/moonratII/pcb_moonratII

It is meant to explain some of that is in the folder.
I just realized that it would be improved by adding an explanatory note and a link to the parts folder.

I would urge any or all of you do make that change so as to become of the habit of building documentation into the project files as we go.   
If it is your first exposure to markdown google and find a tutorial to your liking (and share that tutorial) please.
I keep a markdown syntax cheat sheet taped to a book case above my monitor for quick reference.

Let me know if I can help.
Lee

ForrestErickson, 2024-01-10:
[image]

SilviaKarbid, 2024-01-11:
> From the KiCad schematic there is the properties view of the OLED display [...]

SilviaKarbid, 2024-01-11:
> From the KiCad schematic there is the properties view of the OLED display [...]

Hi Forrest, regarding the display part, the dathasheet we attached is for the oled display that Robert already sent us, actually this issue was to have at hand the dathaseet of the components we have here in Guadalajara. 🌞

SilviaKarbid, 2024-01-11:
> I would urge any or all of you do make that change so as to become of the habit of building documentation into the project files as we go.
> If it is your first exposure to markdown google and find a [...]

> @HJGV05 @SilviaKarbid @kurama79 There is a README.md at the PCB folder here: https://github.com/PubInv/moonrat/tree/main/moonratII/pcb_moonratII
>
> It is meant to explain some of that is in the [...]

ok Forest, I think it would be much better to have explanatory notes, we can work on it as we test the pieces, so we can organize the information a little more.
In the next meting we can talk more about markdown google in order to improve the exiting documentation.

melanielaporte, 2024-02-20:
@melanielaporte Spruce up this weekend.

melanielaporte, 2024-09-18:
@melanielaporte