PubInv/moonrat #198: Add Arduino UNO Stackable Header to JLCPCB BOM

ForrestErickson, 2024-01-27:
A labor intensive activity of the MoonRat II build of 2023 is adding "stackable headers" on the UNO connection pins.

To make the MoonRat II Printed Wiring Assembly more complete, we should add "stackable headers" to the BOM.

A search on JLCPCB for "stackable headers" produced this result with four hits but zero stock
[image]

ForrestErickson, 2024-01-27:
### Parts at JLCPCB but not in stock
**Compare PartDescription Manufacturer Stock  Price** 

PRT-14311
ESP32 Thing Stackable Header Set
Extended Part
Sparkfun Electronics
0
1 + $0.0365 

PRT-09280
SparkFun Accessories Arduino Stackable Header - 6 pin
Extended Part
Sparkfun
0
1 + $0.0365 

PRT-11376
SparkFun Accessories Arduino Stackable Header - 10 Pin
Extended Part
Sparkfun Electronics
0
1 + $0.0365 

PRT-09279
SparkFun Accessories Arduino Stackable Header - 8 Pin
Extended Part
Sparkfun
0
1 + $0.0365

ForrestErickson, 2024-01-27:
Lee sent email to JLCPCB on 20240127

> Hello JLCPCB,
> 
> We have received here in the USA an order from JLCPCB and are very happy so far. I am assisting the person who ordered the assemblies and do not know the order number.  It may be: PCB order review issue to confirm 6875567A_Y3 ----JLCPCB. 
>  
> Our design is for an Arduino UNO Shield we call the MoonRat II.
> 
> We ordered it with out the necessary Stackable Headers required to connect our shield to the Arduino UNO and have locally purchased the stackable headers and are now hand soldering the headers ourselves.
> 
> It occurred to me to check the JLCPCB parts for "Stackable headers" and I found four hits but zero stock.
> 
> 
> When I try to down load the data sheet for "PRT-09280" what I get is a file with a PDF extension but it is HTML and not the part data sheet. 
> I have seen this but for other parts on your web site.
> 
> Please provide data sheets for the following parts:
> PRT-14311  (ESP32 not UNO)
> PRT-09280
> PRT-11376
> PRT-09279
> 
> I see there is zero stock on all of them.
> 
> We would like to build more assemblies soon.
> So that we could build more assemblies but with "stackable headers", Is there a way we can place an order, pay for parts and get them in stock?
> 
> Please advise,
> Lee Erickson
>  
> CC: MoonRat II team members.

ForrestErickson, 2024-01-27:
### Sent message to LCSC components

> Can you provide Stackable Headers like would be used for shields for the Arduino UNO and Due form factor
> 
> Uno Example from Spark Fun is https://www.sparkfun.com/products/11417
> 
> Mega/Due Example from Amazon: https://www.amazon.com/Treedix-Stacking-Headers-Stackable-Compatible/dp/B08G4FGBPQ/ref=sr_1_3
> 
> We would like to order assemblies through JLCPCB and hope you can provide these stackable headers for our future designs.

ForrestErickson, 2024-01-29:
Got reply from JLCPCB

```
01/28/2024 10:12 PM EST JLCPCB Support <[support@jlcpcb.com](mailto:support@jlcpcb.com)> wrote:
 
 
Dear Customer,

Thank you for your inquiry. In general, we recommend that customers choose their own components.

We have a variety of components available for soldering. By entering relevant keywords related to the components you are looking for, you will be able to browse through our extensive catalog and choose the most suitable options based on your requirements. Please visit this link: https://jlcpcb.com/parts.

Should you encounter any difficulties or require further assistance during your search, please do not hesitate to contact us.

Thank you for your understanding.

Best regards,

Kiki
```

Sent the further request.

```
Hello JLCPCB Support,
Thanks for getting back to me.

Regarding, "we recommend that customers choose their own components."
We would like to use parts already in your system so as to not increase your part number complexity.

These are parts in your system most interesting.
PRT-09280
PRT-11376
PRT-09279
Are this parts a mistake ?  I know how hard it is to keep mistakes out of part systems. If that is so tell me please.

For your information, I have tried to contact LCSC components with the same question.
 

However they have not yet replied to my request.
 
So can I choose: Arduino Stackable Header Kit - R3, PN: PRT-11417
https://www.sparkfun.com/products/11417
?
 
Please advise
Thanks,
(Forrest) Lee Erickson
CC: MoonRatII team.
```

ForrestErickson, 2024-01-29:
Email from JLCPCB Support 

```
From: JLCPCB Support <support@jlcpcb.com>
To: Forrest Erickson <forresterickson@comcast.net>
Cc: roy@jlcpcb.com, ml5046@columbia.edu, read.robert@gmail.com
Date: 01/29/2024 7:37 AM EST
Subject: Re: New offline message [LC #12327960.S7IXP2V2NW]
Dear Lee,

Thank you for getting back to us.

Unfortunately, the three parts you are interested in are currently unavailable at JLCPCB.

We apologize for the lack of availability for your request.

For components that are out of stock in JLCPCB or LCSC Parts, please proceed to purchase them from Global Sourcing Parts or submit a request as outlined below. Kindly note that the process may take approximately 9-15 days.

Alternatively, you can choose to replace the out-of-stock components with other feasible options in https://jlcpcb.com/parts.

Thank you.

Best regards,

Kiki﻿
KIKI Chen
Web:	JLCPCB.COM  
Email:	kiki@jlcpcb.com

```

foo
bar

ForrestErickson, 2024-01-29:
Lee sent email to JLCPCB.

> On 01/29/2024 8:43 AM EST FORREST ERICKSON <forresterickson@comcast.net> wrote:
> 
> 
> 
> 
> Hello Ariel and Kiki,
> Thanks for the information.
> 
> We will read the information at the link: https://jlcpcb.com/help/article/53-what-is-jlcpcb-parts-pre-order-service
> 
> 
> 
> Thank you for the drawing to the PRT-14311 part.
> I think I can figure out more from this.
> 
> We are still in the assembly stage and testing for the five assemblies we ordered back in December and have a team meeting Tuesday.  We have identified a change to the PCB and I hope we can add the "Stackable headers" too. 
> 
> You were very helpful.
> 
> 
> Xièxiè
> 
> 
> (Forrest) Lee Erickson
> CC: MoonRatII team.
>

melanielaporte, 2024-03-02:
Removed from v2 because of long lead. Schematic unchanged.