PubInv/moonrat #202: PWA #2 Pin Issue - Smoke

melanielaporte, 2024-02-14:
Pin 12 & 13, has caused burning smell.

HJGV05, 2024-02-19:
Hi Melanie @melanielaporte ! have you solved this problem? have you checked that the other pins have no continuity between them? (I'm catching up with the issues hehe) greetings.

melanielaporte, 2024-02-20:
Hi! Yes, I have just solved the pin issue. At this moment, I am now working on connectivity between your code and the boards. 
Lee has 2 sets of boards as of last weekend and has successfully connected the entire system. We will show our progress tomorrow night. I hope to order PCB version 2.0 very soon so we can begin the build. 

https://github.com/PubInv/moonrat/assets/107901386/a3397e30-9779-48df-a8b8-922d29bdf51d

melanielaporte, 2024-02-20:
@HJGV05 @ForrestErickson Hi! Yes, I have just solved the pin issue. At this moment, I am now working on connectivity between your code and the boards.
Lee has 2 sets of boards as of last weekend and has successfully connected the entire system. We will show our progress tomorrow night. I hope to order PCB version 2.0 very soon so we can begin the build.

https://github.com/PubInv/moonrat/assets/107901386/c6d716ca-b6b0-4860-a24c-6b4cbcac998f

Sound up!

melanielaporte, 2024-02-20:
@ForrestErickson

melanielaporte, 2024-02-20:
Smoke issue solved.

closed 2024-02-20 by melanielaporte