PubInv/moonrat #209: Notes on Assembly of MoonRatII Version 1.0 PWA

ForrestErickson, 2024-02-18:
## Capture notes on the assembly process in an issue as I go.
This can be put into a README.md or other markdown document later.
This content will be good for the Hardware X article.

### Load Blink sketch into IDE
[image]

### Select the COM port
You can see the COM port appear using the Windows Device Manater.
#### Before connecting the USB port
[image]

#### After connecting the USB port
[image]

Under Tools, Port: select the appropriate port.
[image]

Under tools, Board select the Board Manager, Arduino AVR, Arduino Uno
[image]

Select, Get Board Info and capture to **TBD???**
[image]

Press the Upload icon .
[image]

There is a green progress bar in the Status lower right.
[image]

The Status will show Done Uploading when finished.
[image]

ForrestErickson, 2024-02-18:
Each assembly will "enumerate" with a different COM port.   For these MoonRatII assemblies on the factory test computer this will become an opportunity for confusion as the device count increaset.
The first device became COM 10.
The second device became COM12.
[image]

ForrestErickson, 2024-02-18:
Place the MoonRatII shield on the UNO.

**SPECIAL NOTE**
The genuine Arduino UNO has a Type B USB connector which is big enough to interfere with the MoonRatII PCB and short against the switch SW1 pins. 
Trim the pins flush as you can.
With vinyl tape, insulate the pins from touching the USB connector.
The USB connector prevents the MoonRatII from fully seating into the headers.

Future purchases should be of UNOs with the micro USB connector to avoid the possibility of electrical short and mechanical interference.

ForrestErickson, 2024-02-18:
## Check Against Power Short
Measure resistance to ground on power Jack J1
SN 01 19.9K
SN 04 19.9K

Plug into USB cable, power from USB.
The buzzer is buzzing and the amplitude changes with proximity of my fingers.
LED is blinking
No smoke signals.

Plug into wall transformer with current monitor
SN 01 50 mA
SN 04 50 - 70 mA depending on how loud the buzzer is.
No smoke signals.

ForrestErickson, 2024-02-23:
@RobertLRead 
FYI there is an issue into which I was capturing my testing of Version 1.0 PWAs.

ForrestErickson, 2024-03-26:
In March Resume testing.
Lee connected the heater assembly to the SN01 and SN04 with a ribbon cable.
Lee connected the heater pad THROUGH A 100 OHM RESISTOR to the screw terminals at Heater assembly J1. This 100 Ohm resistor is to prevent the assembly from drawing more than the current the USB can supply (100 mA).  

### Run Firmware.
Lee verified that: 
On release of reset the display had the public invention splash screen.
Splash screen was followed by display of firmware version.
Then the SELECT SAMPLE menu came up.

Lee selected the Yeast and mold and pressed SW2 (Enter?)
The display began showing a graph of temperature.

I do not think the temperature is correct.

ForrestErickson, 2024-03-26:
Some of this work should be captured into the README.md article which forms a manufacturing test for Version 1 Assemblies.  
See: 
https://github.com/PubInv/moonrat/blob/main/moonratII/pcb_moonratII/manufacturing_test/README.md

melanielaporte, 2024-09-18:
@melanielaporte