PubInv/moonrat #210: Future Purchase UNO with micro USB connector, Short to SW1 and Interference with Board

ForrestErickson, 2024-02-18:
The genuine Arduino UNO has a Type B USB connector which is big enough to interfere with the MoonRatII PCB and short against the switch SW1 pins.
Trim the pins flush as you can.
With vinyl tape, insulate the pins from touching the USB connector.
The USB connector prevents the MoonRatII from fully seating into the headers.

Future purchases should be of UNOs with the micro USB connector to avoid the possibility of electrical short and mechanical interference.
For example see: https://www.makertechpr.com/product-page/arduino-uno-r3-micro-usb-8-analog-inputs

Modify bill of material.

melanielaporte, 2024-03-02:
Change the schematic to an Uno with smaller micro USB connector.

ForrestErickson, 2024-03-02:
20240301 Changed the Schematic.

closed 2024-03-02 by ForrestErickson