PubInv/moonrat #211: Key Caps Wrong Size For Switch, Version 1 JLCPCB Build

ForrestErickson, 2024-02-18:
The square retention feature of the key caps are way to large for these switches.

Need to look for other JLCPCB part number Key Caps. 

Need to identify and order a correct key cap for SN 1-5.

ForrestErickson, 2024-02-18:
@melanielaporte 
Lets review the data sheets for the switch and the key caps and figure out where I went wrong.

My guess is that one data sheet has key caps for multiple size switches.

melanielaporte, 2024-02-20:
Great, I have time tomorrow before the meeting or Wednesday, both after 5:30pm.

melanielaporte, 2024-02-20:
Ok @ForrestErickson I have time to meet tomorrow or Wednesday, both after 5:30pm

melanielaporte, 2024-02-29:
v2 Order

Keycap number JLCPCB part number: C87037 MFR.Part #
B3F-1052

Update switch cap from B3F-1052 to B32-2000

Digikey - appropriate keycaps d
Mouser...

[image]

ForrestErickson, 2024-03-02:
20240301 Update schematic with Digikey part number.

closed 2024-03-02 by ForrestErickson