PubInv/moonrat #215: Pushing SW1 crashes the UNO, and seems to draw too much current at the same time.

RobertLRead, 2024-02-21:
When I press teh SW1, the system resets. My Mac, which has an intelligent USB, shuts off the USB saying that it drew too much power.

I can see no reason why looking at the schematic. I am trying to understand by studying the PCB.

RobertLRead, 2024-02-21:
Checking with my multimeter, the resistor is indeed a 1K ohm resistor.

RobertLRead, 2024-02-21:
This was solved by adding electricians tape to the large square USB connector. I believe the shield was shorting against it. This is a fairly common problem with Arduino Shields.

I do not know precisely what action sould be taken here. I belive the card should cover that part of the UNO at all, and the barrel jack should be moved to the other side.

melanielaporte, 2024-02-21:
Ok, Lee and I will address the issue this weekend. You are welcome to attend.

ForrestErickson, 2024-02-21:
I too had found this and had been making photos and notes.
A solution will be to purchase UNOs which have the micro USB connector. The offical Arduino UNOs come with the large Type B connector which causes this problem. 
The inexpensive clones can be purchase with the miro USB. 
For example
[image]

ForrestErickson, 2024-03-02:
Updated schematic for the Aliexpress part above.

closed 2024-03-02 by ForrestErickson