PubInv/moonrat #216: ECO, Moonrat II PCB version 1

melanielaporte, 2024-02-21:
For circuit board one: Add a jumper wire from D8 to D10

Change firmware to support DWM D10.

ForrestErickson, 2024-02-21:
### Configuration Records
Record here, by serial number of the MoonRatII Version 1 PCBs for the Jumper D8 to D10 rework.

| **Serial #** 	| **Disposition** 	| **Date of Rework** 	|
|--------------	|-----------------	|--------------------	|
| 1            	| Lee             	| 20240220           	|
| 2            	|                 	|                    	|
| 3            	|                 	|                    	|
| 4            	| Lee             	| 20240220           	|
| 5            	|                 	|                    	|

ForrestErickson, 2024-03-02:
20240301 Update schematic and Version 2 PCB.

closed 2024-03-02 by ForrestErickson