PubInv/moonrat #224: Order Rev 2 PWA and PCBs,

ForrestErickson, 2024-03-02:
## Files to Generate
- [ ] Review Design for Manufacturing JLCPCB
- [ ] Schematic Rev v2 PDF
- [ ] PCB Rev 2 PDF
- [ ] Update the live links to the live viewer (Kicanvas), schematic and PCB at https://github.com/PubInv/moonrat/tree/main/moonratII/pcb_moonratII
- [ ] 
- [ ] Generate BOM (Bill of Material),  Break in to sub BOMs for JLCPCB, DigiKey, Aliexpress, Other.
- [ ] Generate 3D files of PWA (Printed Wiring Assembly) (With and without the UNO?)
- [ ] Export gerber files
- [ ] Review gerber files
- [ ] Order

ForrestErickson, 2024-03-02:
Review for Manufacturier requirements
KiCad before setup
[image]

ForrestErickson, 2024-03-02:
## Verify that the TMP36 is out put values that make sense.
At room temparature of about 73 degrees we had 0.75 V

Here is a graph where about mid way Lee pinched the TMP36 between fingers and the output rise a bit.
[image]

### Conclusion
The TMP36 is wired correctly

melanielaporte, 2024-03-16:
[image]

melanielaporte, 2024-03-16:
@ForrestErickson 
We have changed the Uno type from Uno to R4 Minima. 6 R4 were ordered 20240316 for the Rev 2 prototypes and shipped to GDL team. 
Future plans: order R4 for the US team as they are still working with the old Unos.

[image]
cc: @RobertLRead

closed 2024-09-25 by melanielaporte