PubInv/moonrat #229: KiCAD BOM (Bill Of Material) Required Post Processing for JLCPCB

ForrestErickson, 2024-03-03:
## Summary
KiCAD BOM (Bill Of Material) Required Post Processing for JLCPCB. This proccess was error prone and manual.

### History and Details

When generating a BOM from KICAD for the Rev 1 PCB from KiCAD significant effort and several submissions of the BOM to JLCPCB took place as we tried to understand and work to the requirements of JLCPCB automated input.

In particular their system did not like "MF4-MF6, MF8" and wanted instead "MF4-6, MF8" or "MF4, MF5, MF6, MF8"

### Idea to try, Parts List
A parts list has one record per reference designator
By changing the grouping when using the Symbol Fields Table to generate a CSV list we get in stead of a BOM a Parts List

[image]

ForrestErickson, 2024-03-03:
Example generated as above

[image]

melanielaporte, 2024-12-05:
@ForrestErickson I created a JLCPCB order repo. Please review and let me know your thoughts. I will add images soon. https://github.com/melanielaporte/JLCPCB-Ordering-Guide/

closed 2024-12-05 by melanielaporte