PubInv/moonrat #234: Rev 2 MoonratII Boards and Assemblies Order Information Capture 20240307

melanielaporte, 2024-03-07:
1.Our previous order #W20231130095 required an additional $25.29 for the v-cut.
This order, we have selected 2 boards, Paneled by Customer. 
We would like the PCBs and assemblies to be similar to our previous order 
2. MF12 (1x4 header) should be soldered to J2 (OLED display) 
3. MF 4-6, 8 (round nuts) mounted to OLED display 
    
If additional payment required, please advise.

melanielaporte, 2024-03-07:
Things necessary to make a successful order not obvious to a beginner 
Uploading the .... files, Selected Research development
Friday - do a dummy order, screen capture each step @melanielaporte

melanielaporte, 2024-03-16:
JLCPCB email
[image]

melanielaporte, 2024-03-16:
Our response
[image]
[image]

melanielaporte, 2024-03-16:
JLCPCB email 20240316
[image]

melanielaporte, 2024-03-16:
@ForrestErickson

melanielaporte, 2024-03-21:
Issue with JLCPCB order: due to the PCB being ordered Economy instead of Standard board, added the v-cut is not possible. 

@ForrestErickson There are 2 options. Which should we choose?
1. Go forward with the Economy PCBA Rev 2 order, requiring the GDL team to cut the board manually.
2. Cancel the Economy {CBA Rev 2 order and purchased a Standard Rev 2 PCBA today April 21st.  
[image]

melanielaporte, 2024-03-23:
Doing it right now, will keep the team informed.

melanielaporte, 2024-03-23:
@ForrestErickson The site is down tonight, I will order in the morning, 03/23/2024.
[image]

melanielaporte, 2024-03-24:
New PCBA order has been placed, invoice sent.

[image]
@RobertLRead @ForrestErickson

ForrestErickson, 2024-03-25:
@melanielaporte 
Can you summarized here the cost delta so we remember? IIRC it was not much but we should know that the "V" cut is a distinct cost in addition to the component and board and assembly costs. 
Can you please  itemized this cost in the Hardware X article too?

If we had plenty of free time we could even make a copy of the MoonRatII schematic and PCB and break into two schematics and two PCBs. One each for the Controller and the Heater assembly and then work up an order with JLCPCB for the two assemblies seperate. Another order with the two assemblies "Panelized" to further understand the cost implications. 

CC: @RobertLRead

melanielaporte, 2024-03-26:
JLCPCB production is finally a GO!

[image]

closed 2024-09-25 by melanielaporte