PubInv/moonrat #237: Ship PWA to Harshit, enclosure design

melanielaporte, 2024-03-11:


melanielaporte, 2024-03-11:
@ForrestErickson Should I include a raw board and all components?

ForrestErickson, 2024-03-12:
Regarding, " Should I include a raw board ..."
Yes because it may be useful. 

Regarding, "and all components?"  I am not sure what you mean.
Do you mean, " and a full printed wiring  assembly?".  The answer is yes. because testing fit for an enclosure needs components on the board to check for fit to the display and buttons and buzzer and against interference with all of the connectors.  

And we should be examining our assumptions.  Will our enclosure designer have everything needed for representative testing?  Correct key caps is an important part of that but not the only part.
For the Rev 2 PWAs the big transistor Q1 has moved slightly to the left relative to the Rev 1 PWA.  

Finally, if you can, You should solder a wire onto the PCB to improve the current carrying capability on the VCC net.
I will try to do so some time today but I have several zoom meetings and a telescope to the library activity this afternoon competing for my time.

Do you have some wire with which you could try to improve the connection from the down current side of the fuse, F1 to a connection near J6 pin 2?
This will require you to scratch off the green PCB coating called solder resist to solder to the surface of the copper trace near J6 pin 2.  A great thing to learn how to do and not obvious.

@harshit9602 Do you have the ability and tools to solder wire to PCB traces?

melanielaporte, 2024-03-16:
@ForrestErickson Soldered wire to PCB traces for @harshit9602's Rev 2 PWA set.

closed 2024-03-16 by melanielaporte