PubInv/moonrat #238: Screw distributor issue

SilviaKarbid, 2024-03-21:
Put here screw distributors in Guadalajara 

[image]

SilviaKarbid, 2024-03-21:
I found this, and then have a place here in Guadalajara to pick the screw
https://www.torec.mx/products/tqp2x6
[image]
[image]

SilviaKarbid, 2024-03-21:
They accept paypal

ForrestErickson, 2024-03-21:
The screw type should be pan head which has a flat face against the OLED PCB.

@SilviaKarbid 
Can you find one meeting those requirements?

SilviaKarbid, 2024-03-21:
I will check 🙂

closed 2024-09-25 by melanielaporte