PubInv/moonrat #239: Cell Phone Interference with Temprature Measurement, Rev 1 Assemblies

ForrestErickson, 2024-03-26:
Lee discovered that placing a cell phone near the cable connecting the controller and heater causes the measured temperature to be in error.  
[image]

RobertLRead, 2024-03-26:
Interesting. Can this be fixed with shielding?  Should one move to a
digital temperature transmission?

ForrestErickson, 2024-03-26:
Regarding, " Can this be fixed with shielding?"

This morning I am not able to reproduce the problem.  

No idea if the cell phone operation is some how different or something else.
The wires on my work bench are a rat's nest. I have not moved them much but some movement has happened with out a doubt. 
Not sure where my arms need to be for example.

Just not sure what do try to make further testing and make the problem repeatable.

ForrestErickson, 2024-03-26:
I have configured my oscilloscope to measure the AC waveform on the A0 input.
Because it is a small signal the ground bounce (interference may be non trivial so I made a two channel differential measurement.

The red (Math) channel is A0 - Ground and so the ground hum (interference) is removed.
[image]

There are occasionally much greater bursts of interference than I captured here.

ForrestErickson, 2024-03-27:
## More investigation.
I made a phone call and observed interference.

Interference in measurement
[image]

Oscilloscope waveform. NOTE the vertical scale is significantly different on this measurement.
[image]

RobertLRead, 2024-09-25:
This MAY be irrelevant now that we have moved to using a digital thermometer. However, we should retest, so I am leaving this open.

closed 2025-08-29 by melanielaporte