PubInv/moonrat #243: Wires problem

SilviaKarbid, 2024-04-16:
We have doubts about where to connect the cables we received because they are too large to connect to the PCB.
1. Are the screws in the image for the edge of the PCB or for the screen? 
2. If they are for the screen, they seem a bit small.
3.The pins to solder to the PCB are missing. Do they come in the missing package or would you like us to purchase them here?
[image]
[image]

RobertLRead, 2024-04-16:
I think @ForrestErickson or @melanielaporte will have to answer this.  I assume we can discuss it tonight.

RobertLRead, 2024-04-16:
recall that the idea is that the cable be cut in half, and the presumably the cut halfs connected to the board (one side) and the heater appliance (on the other), with the male and female ends becoming a means of coupling and decoupling them.  Is it possible that the screw-clamp connector on the main board is meant to serve that purpose?

ForrestErickson, 2024-04-17:
## Choice of which cable 
I remind everyone that the automotive lighting cable can be used OR the flat ribbon cable may be used but we should chose one or the other. They each have positive and negative consequences for ease of assembly and the strain relief which must be designed into the enclosure.
I assert we must fabricate and evaluate a design for each type of connection.

### To use the automotive lighting cable
The cable is to be cut in half. The wires may be too thick for all of the strands to fit into the screw terminals. I have not tried it  yet. If you can try let us know. 
Some strands must be sacrificed till what remains fits.

Because power will be coming from the control assembly, it is important that the half of the cable with this end (pictured)
[image]
the end where all but one terminal is surrounded by insulation, be wired to the control board such that ground is the single exposed pin.  This is so that power can not be shorted  due to an accidental touch of the cable to anything.

The rest of the wiring must simply be the same on both the control assembly and the heater assembly. 
The wiring must be documented so that we wire every single control assembly and heater assembly the same so that they are all interchangeable. 

The other end of the cable with the five exposed contact pins is wired to the heater assembly.

## A "To Do"
Document the wiring when using the automotive lighting cable.

ForrestErickson, 2024-04-17:
Regarding, @harshit9602  statement, "Are the screws in the image for the edge of the PCB or for the screen?
If they are for the screen, they seem a bit small."

The intent is for screws to attach the screen. Please try them and get back to us on the fit.

ForrestErickson, 2024-04-17:
Regarding, @harshit9602 statement, "3.The pins to solder to the PCB are missing. Do they come in the missing package or would you like us to purchase them here?"

There are supposed to be headers with long pins soldered where indicated here.
[image]
@melanielaporte can you identify the part number and the Amazon link for purchasing these parts.

**Problem?**  IIRC Harshit does not have equipment with which to solder?

SilviaKarbid, 2024-04-18:
> The intent is for screws to attach the screen. Please try them and get back to us on the fit.

The screws are thin and don't fit the screen, but since the screen is soldered to the board, I think for now it can be left like this

ForrestErickson, 2024-04-18:
Hello @harshit9602 ,
Can you measure the OD of the screw thread?
Can you measure the ID of the threaded insert?

For the threaded insert, here is the data sheet: https://wmsc.lcsc.com/wmsc/upload/file/pdf/v2/lcsc/2205251700_Sinhoo-TH162050CTJ_C3020130.pdf

From the drawing I infer a 2M screw should work. See detail here:
[image]

Please advise.

harshit9602, 2024-04-18:
I dont think I follow what you are trying to ask. My approach right now is to design the enclosure that fits everything and then find a screw that would hold it in place and not the other way around. Also, i dont have calipers to measure OD and ID of screws

ForrestErickson, 2024-04-21:
@harshit9602 
Regarding, "I dont think I follow what you are trying to ask."
I was trying to figure out if the screw is wrong or the mounting hole is wrong. The intent is for both to be 2M.
Of course something else may be wrong and we are miscommunicating.
A photo would help.

Thanks,

ForrestErickson, 2024-05-10:
Regarding "We have doubts about where to connect the cables we received because they are too large to connect to the PCB."

The screw terminals are on a 0.1" spacing.
A larger screw terminal on a 0.156" spacing may work better for the heavy gauge wire, but of course is a PCB redesign.

Here is an example of such a screw terminal:
https://www.aliexpress.us/item/3256804473999312.html?spm=a2g0o.productlist.main.9.42a53076rVeo50&algo_pvid=8edf037b-9219-4843-8d86-fdd6810ac1e0&algo_exp_id=8edf037b-9219-4843-8d86-fdd6810ac1e0-4&pdp_npi=4%40dis%21USD%211.94%211.94%21%21%211.94%211.94%21%402101e58317153374751157829e2d5b%2112000030016018201%21sea%21US%212057004976%21&curPageLogUid=tz8Hy3OywGLw&utparam-url=scene%3Asearch%7Cquery_from%3A

From JLCPCB there are 5mm pitch screw terminals. Example:
https://jlcpcb.com/partdetail/Jiln-JL50050006G01/C2802619

melanielaporte, 2024-09-18:
@ForrestErickson Close?

ForrestErickson, 2024-09-19:
I think until we have an enclosure or enclosures fully designed we will remain uncertain on the wires which connect the controller to the incubator (heater)

RobertLRead, 2024-09-25:
We now know to increase size of screw clamp connectors for the automotive brake light connectors.

RobertLRead, 2024-09-25:
At present, the GND connection of the digital thermometer has not obvious place to be connected. In a future design, the resistor should be on the PCB (NOT a flying solder joint). Also, a connector for GND is needed. At present, the screw clamp connector could be reused for GND, but it would be hard to connect two wires here.

melanielaporte, 2025-08-29:
@RobertLRead Will test today.

closed 2025-11-03 by melanielaporte