PubInv/moonrat #244: Document wire color of automotive cable harness convention for connection between controller and sensor PCBS

RobertLRead, 2024-04-24:
Please put this manufacturing assembly documentation.

SilviaKarbid, 2024-05-07:
[image]

SilviaKarbid, 2024-05-07:
[image]
[image]

ForrestErickson, 2024-05-08:
Thanks.
This needs to become part of the HardwareX article soon.

I confess that I have had zero learning about Latex editing since our first meeting regarding editing in Latex and so cannot do this at this time.

Lee

closed 2025-11-03 by melanielaporte