PubInv/moonrat #264: Prusa Slicer Returns Warning on case_incubator.FCStd,

ForrestErickson, 2024-05-15:
Lee loaded the FreeCad file:
https://github.com/PubInv/moonrat/blob/main/moonratII/enclosure_mooonratII/heater_assembly/case_incubater.FCStd

Lee exported an STL file.
In preparation to print, Lee loaded the STL file into PrusaSlicer.
The slicer returned warnings. 
[image]

The design will require "supports" because there is a large horizontal surface not on the build plate.

Lee will enable supports and try to print.

ForrestErickson, 2024-05-15:
## Notes on trying to set up PrusaSlicer for support.

[image]

melanielaporte, 2025-08-28:
Solved, 202508.

[image]