PubInv/moonrat #270: No Render, "Perforated_plate.scad",

ForrestErickson, 2024-05-23:
Rob,
I keep learning more about the Moonrat past.
During our zoom with Auja you showed the perforated platform on legs (the spindly ones) which was new to me.
I went looking and found in the folder now under legacy some SCAD files including "Perforated_plate.scad"
[image]

However when I open it in my SCAD, I get no render and a warning
"WARNING: No top level geometry to render "
[image]

Can you duplicate the problem? Is there a simple fix on my end?

RobertLRead, 2024-05-23:
Forgive me; I was working on that in OpenSCAD, and then I think @harshit9602 took over and designed the perforated plates. I don't know if he ever committed that design file. (The legs are too weak in any case.)

I could duplicate the work quickly in OpenSCAD if it become necessary (Harshit may have used a different tool.)  I will assign this to Harshit to have him check in the file (if he still has it).

ForrestErickson, 2024-05-29:
Robert wrote the following email:
Dear Lee, 
While in the  meeting with xxxxx, I updated the code here:
https://github.com/PubInv/moonrat/blob/main/legacy/moonrat/Perforated_plate.scad
The round perforated plate is 120mm in diameter. The legs make the disk about 13mm high. (The legs are too fragile, BTW.)
The current OpenSCAD does not add legs. I still think Harshit made the parts I actually have. But I could quickly make them OpenSCAD if I have to.

Lee updated his fork and confirms that the part now renders in OpenSCAD.
[image]

Closing this issue.

closed 2024-05-29 by ForrestErickson

harshit9602, 2024-06-16:
Hey Robert,
Sorry for being away for so long. I was out on a company trip to Ohio on a
legal project. I am attaching the files that we can print and see if things
assemble as per need or not. if it needs more adjustment, i can do that.
For now someone else needs to print this and verify. I think Lee has a 3d
printer. Really sorry for the trouble

harshit9602, 2024-06-17:
Harshit,
 
Please note change of subject line on this email.
 
Thanks for the files. I have placed the files into the repository at:
https://github.com/PubInv/moonrat/tree/main/moonratII/enclosure_mooonratII/Control_Enclosure_FreeCAD
 
 
I have opened the FreeCAD file and find that this is an enclosure top.
I have renamed the files to make the names more descriptive.
 
I see a mounting bos only for two of the four mounting locations of the PCB.
All four mounting holes should have screws.
Please also provide all original files. For example, If this is done in Solid Works please include those files.
 
It would be best if all files were under version control at the repository.
Do you need help learning the Git and Github and work flow for file version control?
 
Lee

ForrestErickson, 2024-06-17:
This should have been the subject line: MoonratII_Enclosure_Controler_top