PubInv/moonrat #276: Error: 'AR_DEFAULT' was not declared in this scope, When compiling in Arduino IDE

ForrestErickson, 2024-06-11:
Lee got a compile time error when compiling the code added with merge: Sensor proofs code #275

This screen shot shows the Arduino IDE settings and the error returned by the compiler
[image]

Code file name: sensor_proof.ino

Hardware under test:  MoonratII Version 1 with an R3 UNO.

kurama79, 2024-06-11:
Hi Lee,
The problem is that the uncommented code is for testing the temperature sensor with an Arduino R4. If you want to test it with the R3, you need to comment out those four lines and uncomment the lines that follow the line saying "Use this part for R3."

ForrestErickson, 2024-06-11:
Ok I think I see now. THe comments make it clearer once I understood.

[image]

it appears that lines 26 through 31 are for R4 while lines 33 to 47 are for R3 UNOs.

I will investigate...

ForrestErickson, 2024-06-11:
I have no R4 with which to test.

I have created some #ifdef with which to manage the controller dependent code.
I changed the serial print formatting so that the three reported results come out in a block 
I have compiled and loaded on an UNO R3.

Screen shot of running code on an R3 MoonRatII Version 1 assembly
[image]

ForrestErickson, 2024-06-11:
I have committed some changes and hope @HJGV05  and or @kurama79 can try out on the UNO R4 
You will have to comment out the line 7 and uncomment line 8

[image]

ForrestErickson, 2024-06-12:
Regarding, 

> "Hi Lee,
> The problem is that the uncommented code is for testing the temperature sensor with an Arduino R4. If you want to test it with the R3, you need to comment out those four lines and uncomment the lines that follow the line saying "Use this part for R3.""
> 

I should also have asked, Why does it not compile for the R4 on my IDE. I am not loading it into the R3 just compiling. Is there a setting on the IDE I need to make?