PubInv/moonrat #278: Lee wants to Order UNO R4, Which One?  WiFi or Minimua or ???

ForrestErickson, 2024-06-11:
Hello team,

Lee wanted to order an UNO R4 but discovered on Amazon that there is no such thing. There are several.
[image]

Which R4 do the users in Guadalajara have?

SilviaKarbid, 2024-06-11:
Hi Lee!, we use this one, but we think that maybe the WiFi module make some interference with the voltage, I don't know if you want to try the R4 minima [image]

ForrestErickson, 2024-06-12:
FYI,
I have placed an order for one and will soon have it.
[image]

RobertLRead, 2024-07-30:
I have two R4 minimas. I plan to start building soon; hopefully we will have a confirmation that that works soon.

RobertLRead, 2024-09-25:
@ForrestErickson : We are using R4 Minimas; that is what you should order, I am closing this issue.

closed 2024-09-25 by RobertLRead