PubInv/moonrat #288: HardwareX professional headshot

melanielaporte, 2024-07-17:
In Discord or ml5046@columbia.edu: send Melanie a professional headshot (do not include body)

ForrestErickson, 2024-07-19:
I know I have and suspect other volunteers have a photo on the Public Invention org web page.
See:  https://www.pubinv.org/inventors/

The photo there is the one to be used for me.

melanielaporte, 2024-07-19:
Thanks Lee, the paper still requires a 100 word or less biography from you. I assume your affiliation is Public Invention. Add your biography below in a comment

ForrestErickson, 2024-07-19:
(Forest) Lee Erickson
Public Inventor of the General Purpose Alarm Device (GPAD) Project
B.S. Physicist University of Minnesota.
I am the luckiest person to have ever lived because when I was ten years old, I was taken to the moon by Michael Collins, Buzz Aldrin and Neal Armstrong (as were about five billion other persons watching television.)

I have had a career doing elementary analog circuit design for audio, video and deflection systems and embedded system with some focus on electro- magnetic compatibility. I worked for a high volume consumer electronic company where I learned the importance of design for manufacturing and project management of medium- to large-multidisciplinary teams. I also ran my own company for 12 years where I had to take on almost all roles from time to time.

With Public Invention I have helped with the General Purpose Alarm Device (GPAD) project especially with Schematic Capture and PCB Design and testing. I wrote and simulated firmware for factory testing of the Version 1 PCB. From my Public Invention involvement, I have improved my understanding of how to use GitHub (and GitLab) for managing product development from my team mates. My short term interests are to document well the GPAD as a template to help Public Invention and others.

I am a refugee from the winters of Minnesota and live in East Tennessee where snow is sufficiently infrequent that it is mostly received as decorative when it happens. My supper power is NOVEL SPELLING ( and diverse spelling). My Github is found at: https://github.com/ForrestErickson

melanielaporte, 2024-07-23:
Your 100 word biography: I am the luckiest person to have ever lived because when I was ten years old, I was taken to the moon by Michael Collins, Buzz Aldrin and Neal Armstrong (as were about five billion other persons watching television.)

I have had a career doing elementary analog circuit design for audio, video and deflection systems and embedded system with some focus on electro- magnetic compatibility. I worked for a high volume consumer electronic company where I learned the importance of design for manufacturing and project management of medium- to large-multidisciplinary teams. I also ran my own company for 12 years

closed 2024-07-23 by melanielaporte