PubInv/moonrat #298: Create Assembly guide - please take excellent photos!

melanielaporte, 2024-07-17:
Create assembly guide of every step to create the prototype, from drilling the hole to splitting and attaching cables.

SilviaKarbid, 2024-07-30:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

RobertLRead, 2024-07-30:
> WhatsApp.Image.2024-07-29.at.7.20.28.PM.1.jpeg (view on web)
> WhatsApp.Image.2024-07-29.at.7.20.28.PM.jpeg (view on web)
> WhatsApp.Image.2024-07-29.at.7.20.27.PM.jpeg (view on web)
> WhatsApp.Image.2024-07-29.at.7.20.26.PM.jpeg (view on web)
> WhatsApp.Image.2024-07-29.at.7.20.25.PM.1.jpeg (view on web)
> WhatsApp.Image.2024-07-29.at.7.20.25.PM.jpeg (view on web)
> WhatsApp.Image.2024-07-29.at.7.20.24.PM.1.jpeg (view on web)
> WhatsApp.Image.2024-07-29.at.7.20.24.PM.jpeg (view on web)
> WhatsApp.Image.2024-07-29.at.7.20.23.PM.1.jpeg (view on web)
> WhatsApp.Image.2024-07-29.at.7.20.23.PM.jpeg (view on web)
> WhatsApp.Image.2024-07-29.at.7.20.22.PM.1.jpeg (view on web)
> WhatsApp.Image.2024-07-29.at.7.20.22.PM.jpeg (view on web)
> WhatsApp.Image.2024-07-29.at.7.20.20.PM.jpeg (view on web)

Bravo!  Well done!

melanielaporte, 2024-07-30:
Great photos!

RobertLRead, 2024-07-31:
Trying to get this into overleaf:

[image]
[image]
[image]
[image]

RobertLRead, 2024-07-31:
I have added all of these images to the Overleaf project, and put \label tags in them so they can be referred to. I have not attempted to describe them yet.

I'm going to leave this issue open as there are yet more photos that we may want to make. @SilviaKarbid

melanielaporte, 2024-09-18:
@RobertLRead

RobertLRead, 2024-09-25:
@SilviaKarbid please add photos of digital sensor and resistor soldering and THEN close this issue.

melanielaporte, 2025-08-27:
@melanielaporte 20250828