PubInv/moonrat #299: Find similar paper references = better chance to be published

melanielaporte, 2024-07-17:
_References
If relevant, you should include a reference to the original publication of the hardware you customized and a
reference to the repository in which your design files are published. Other references can be included, as required;
for example, references that put your device in context in the literature. For more information on the reference
format in HardwareX please see the https://www.sciencedirect.com/journal/hardwarex/publish/guide-for-authors_

melanielaporte, 2024-09-18:
@RobertLRead @melanielaporte @ForrestErickson

SilviaKarbid, 2024-09-25:
Hi, for fuzzy logic controller book -> introduction to fuzzy sets, fuzzy logic, and fuzzy control Systems, Autors : Guanrong Chen, Trung Tat Pham, pages 89-120

RobertLRead, 2024-09-25:
@HJGV05 is going to place more Fuzzy Set/Logic references here.

HJGV05, 2024-09-30:
Thank you @RobertLRead  and @melanielaporte. Here are the suggested references for the fuzzy logic controller:

Chapter of book: https://books.google.com.mx/books?hl=es&lr=&id=yNqgDwAAQBAJ&oi=fnd&pg=PA325&dq=fuzzy+logic+mamdani&ots=c3gXuxgrnT&sig=KO8nzs4eJMAEjKMvmC-UGSIBXxo&redir_esc=y#v=onepage&q=fuzzy%20logic%20mamdani&f=false

Relevant and related papers:

https://ieeexplore.ieee.org/abstract/document/7065721?casa_token=tsbamUZnFSsAAAAA:et9wI8H1Hw3KsqPDZTkbIdJzMQCsnn1vmRVdOwFOImIRqGHog7M3UJxFR_XB2caABuX9__Ch46CS

https://ieeexplore.ieee.org/abstract/document/5230275?casa_token=ga32WBr0N_4AAAAA:sbX_5W9YtBkIyMRbLXw1dLvJee-Rk8-2VF0G8HxWqsfZCtinvuLvOF2xuSpaDBFQHnu50xt9aL5S

https://ieeexplore.ieee.org/abstract/document/8559861?casa_token=OtyHFU3NF5YAAAAA:yybueE5AgSnGwz4gSb_PP7EZGlOCKSJzi1nepbUcU_iX0dAynzRA65SLVNIDvRDZaVszY9kjSpuX

https://d1wqtxts1xzle7.cloudfront.net/53229408/4-libre.pdf?1495450763=&response-content-disposition=inline%3B+filename%3DControl_of_the_Heating_System_with_Fuzzy.pdf&Expires=1727663898&Signature=NqgycPC7cpZK5IKbtjltdiKrP2EgCG9IHJoMNtXJ2iTBnxlVW9tJ6rCkYdSqmzaNVnv75jI9zCRUWZQApIfdkHbINAEpFK1VY3x5ql7uzhQbEstS7xxdOgBizvIBLfUpMnyTwqfI0hoO2ApNsoR9cmGGk4C1nf7mAYID3hoCW~PghM6ePOHfDXA5npo3MHCGsDhp1NWqxNgwdQ7wGAp2E7pKV6bURV7jMCGwWDZzBov7JKrsH33ljUvi-hgEj5lZPdSBwiuh6ryrACZHy89zKePGoTCOu16-cYD-S67azNA91Z2mhngc06YnVe3z3Yhp8jb7GaL9ahH0WxYtce-bbg__&Key-Pair-Id=APKAJLOHF5GGSLRBV4ZA

I am looking forward to any additional questions or details, best regards.