PubInv/moonrat #301: Send Dr. Read lithium ion battery link for purchase

melanielaporte, 2024-07-17:


SilviaKarbid, 2024-07-25:
I regret to inform you that I did not find the batteries. I searched in the various places where packages are usually left, and I was told that they were not found. I don't know if a claim can be filed with Amazon, or how you would like to proceed. If not, I can provide you with another address for the shipment.

RobertLRead, 2024-07-25:
Give me another address.  I will complain to Amazon, but we should not that
delay getting the batteries to you.

SilviaKarbid, 2024-07-25:
Ok, the address is: 
Corona Boreal 3980
Col. Arboledas 1 sección 
Zapopan Jalisco
postal code: 45070
We prefer that only one battery be sent because we are all working on the same prototype, and this way it won't be as expensive.

RobertLRead, 2024-07-30:
This has been ordered.

closed 2024-09-18 by RobertLRead