PubInv/moonrat #302: Professional portrait/biography/affiliation

melanielaporte, 2024-07-17:
The following information is needed and will be published in the paper. 
Please comment below.
**Note: all included will be publicly visible 
- 1 professional headshot (do not include body)
- professional biography, 100 word maximum
- university affiliation

SilviaKarbid, 2024-07-25:
Silvia Aurora Casillas Romero is a PhD student in Electrical Engineering. Her current field of study is artificial intelligence, and she has a solid background in programming, frequency filter design, 3D modeling, and artificial intelligence. Silvia holds a Master's degree in Electrical Engineering, during which she worked on projects related to hydroponics. She is currently interested in deepening her knowledge of the development of artificial intelligence systems.
University aff: Cinvestav Unidad Guadalajara

kurama79, 2024-07-26:
L. Enrique Ruiz-Fernandez is an AI developer specializing in robotics. He is currently completing his Ph.D. in Robotics, holds a Master's degree in Electrical Engineering, and a Bachelor's degree in Control Engineering. His research focuses on autonomous navigation and collision avoidance for drones. With a strong background in control theory and electronics, he creates innovative solutions for complex engineering challenges. Additionally, he teaches key engineering courses, sharing his expertise with future engineers.
University aff: Cinvestav Unidad Guadalajara and Universidad de Guadalajara

closed 2024-09-18 by melanielaporte