PubInv/moonrat #304: Development and Management of WOKWI Simulations for MoonRatII

ForrestErickson, 2024-08-09:
# Summary
A WOKWI Simulation or Simulations of the MoonRatII Version 2 PCB wiring my be of great value

## Request
Have any of you created a schematic in WOKWI that has good fidelity to the MoonRatII Version 2 PCB ?
Lee suspects there may be more than one.
Can we share this work soon so? I hope to help Robert with the troubleshooting he is doing right now on a unit he hopes to take into the field very soon?

If there is already in this repository a WOKWI simulation please help me find it.

Thanks,
Lee Erickson

HJGV05, 2024-08-09:
Hello Mr Lee!

I hope you are fine, sure, here is the last Wokwi project we created: https://wokwi.com/projects/390933163302682625

 It was attached to the script versions, but since we used some libraries that are not available in Wokwi (like the eFLL library) we did not continue updating it. I hope it helps!

Best Regards,

Horacio García

ForrestErickson, 2024-08-09:
@HJGV05 
It appears that the EFLL libaray can be added through the WOKWI library manager.
[image]

and
[image]

ForrestErickson, 2024-08-09:
I have copied the most current code into a new WOKWI at:
https://wokwi.com/projects/405744143993619457

It fails to compile:
[image]

I assume this is due to the UNO Version 4 but am not certain.

RobertLRead, 2024-08-09:
I made some changes as defined below, which allow it do compile; I interpret this as being either WOKWI does not today support the R4 at all, or we have not properly configured it to do so.  (Please observed commented-out lines below!
I do not assume this works; I just got it to compile. @ForrestErickson 

`#elif defined(R4) //Configuration for Arduino UNO R4
 // analogReference(AR_DEFAULT);  // Internal reference
 analogReference(DEFAULT);  // Internal reference
  delay(10);  // Awaiting reference stability
  // float sensorValue = map(analogReference(), 0, 5.0, 0, 255.0);  // Internal reference mapping
  float sensorValue = map(DEFAULT, 0, 5.0, 0, 255.0);  // Internal reference mapping
  long vcc = (1000 * 1.1 * 1023) / sensorValue;

#else
#error Serial.println("Unsupported board selection.");
#endif
  return vcc;
}

void setup() {

#if defined(R4)
//  analogReadResolution(10);
//  analogReference(AR_DEFAULT);
#else`

closed 2025-08-27 by melanielaporte