PubInv/moonrat #305: Arduino R3 is incapable of running current Fuzzy Logic code

RobertLRead, 2024-08-10:
Probably because of memory.

It is very clear that just by uncommenting certain code, which is not executed until AFTER the OLED is initiated, we can prevent the OLED from initiating. This is code that allocates the "Fuzzy" objects.

I conclude that the Fuzzy logic will run on an R4, but not on an R3.

However, that analog temperature read basically doesn't work on an R4. (It is not clear to me that it works on an R3.)

So we are stuck.  I am going to implement a PID controller on an R3; I believe there is a high probability that that will work, but I will check all the hardware features (esp. the temperature read) first.

ForrestErickson, 2024-08-11:
## Lee experimenting with Wokwi simulating the MoonRatIi on an R3
See: https://wokwi.com/projects/405744143993619457

I am worried I am not using the same code as is Robert.

## Simulation Compiles with Fuzzy Logic, OLED begin error.
At run time I get an error "Error while initializing OLED"
Screen shot
[image]

## Notes on code changes to R3
In the code I have, I have changed with in the code the #define for R3 

> // Set controller type here
> #define R3 //Uncomment for Arduino UNO R3
> //#define R4 //Uncomment for Arduino UNO R4
> 

When I do so the code compiles so I conclude that running the Fuzzy Logic code is possible.

I have compared and contrasted the code which I believe to be inializing the OLED and do not find differences against examples which compile and run.  
Here is one such example: https://wokwi.com/projects/405917918926102529

melanielaporte, 2024-09-25:
We are no longer using the R3 and have moved to the R4.

ForrestErickson, 2024-10-31:
# Reopening and Trying to Duplicate Problem in Maryville

## Journal of Testing
Lee has installed the eFILL library
[image]

Lee has found example codes from the eFILL library
[image]

Lee has set the IDE for UNO and opened and compiled the example code from the eFILL library
[image]

The compiler reports:
> Sketch uses 11482 bytes (35%) of program storage space. Maximum is 32256 bytes.
> Global variables use 264 bytes (12%) of dynamic memory, leaving 1784 bytes for local variables. Maximum is 2048 bytes.

## Conclusion
Lee is unable to duplicate the problem "Arduino R3 is incapable of running current Fuzzy Logic code".
Their may be insufficient description of the setup where Robert (and others?) had the error.

RobertLRead, 2024-10-31:
Lee is right. I have tried to compile both Horacios most recent code and mine, and both seem to load into the R3.

I can't explain if my creation of this issue was in error was based on some legitimate reason that I have forgotten. I am closing this issue for now.

closed 2024-10-31 by RobertLRead