PubInv/moonrat #306: TMP36 output voltage steady in the presence of reference voltage sags....

RobertLRead, 2024-08-11:
When the MoonratII is power by a USB cable but connected to a 12V battery with a switch, switching the battery on and OFF does NOT change the output of the TMP36 voltage as measured by an oscilloscope on an Arduino R3.

However, using both a voltmeter and an oscilloscope, the 5V pin of the Arduino R3 voltage changes from 4.66 V to 4.99 volts. This has an immediate effect on the digitally measured temperatre, as evidenced by the screen shot below, which shows the voltage being switch off. 
[image]
Note the ratios 4.99/4.66 and 166/156 are the same (1.070) to the third decimal place, confirming @ForrestErickson 's statement that the ADC converted measures a ratio.

We may futhermore conclude that this is only correct if the 5V voltage does not sag due to a load; however, when the system is powered at 12V, this is approximately correct. 

Although this fragility is distressing, it may not matter if we power the incubator with a 12V power supply.

ForrestErickson, 2024-08-11:
While I assume you are using the R3 Arduino you should state so.
Please also state the exact firmware used.

Then, the first think I am going to do is look at the voltage on the reference to be set for 1.1V.
From the documentation

> INTERNAL: a built-in reference, equal to 1.1 volts on the ATmega168 or ATmega328P and 2.56 volts on the ATmega32U4 and ATmega8 (not available on the Arduino Mega)

This will require a gain change for calculating the temperature.

ForrestErickson, 2024-08-11:
By the way I see no firmware in the folder where I think it is supposed to be:
https://github.com/PubInv/moonrat/tree/main/moonratII/firmware_moonratII/production_R4/MoonTestV5
[image]

RobertLRead, 2024-08-11:
My new R3 work is (and will be) here:

https://github.com/PubInv/moonrat/tree/main/moonratII/firmware_moonratII/production/moonratIIR3Rob

Being uncertain of myself, I put my latest R4 code here:

https://github.com/PubInv/moonrat/blob/main/moonratII/firmware_moonratII/production_R4/MoonRatRob6/MoonRatRob6.ino

However, I am now focuing on the R3, as it gives me some hope of building a system that I can take to Tanzania.

melanielaporte, 2024-12-07:
This issue is irrelevant since we are using the R4, correct? @RobertLRead

closed 2025-08-27 by melanielaporte