PubInv/moonrat #311: Change for DS18b20 Digital Temperature Sensor

ForrestErickson, 2024-09-18:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
The analog sensor is a victim of interference and is unusable on the Arduino R4.
Migrate the design from the analog temperature sensor to a digital sensor

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
Change for DS18b20 Digital Temperature Sensor

SilviaKarbid, 2024-09-25:
Hi Lee, we added the DS18b20 sensor. The sensor has a 4.7k resistor between the signal and the voltage, and it was soldered directly to the signal output and the 5V power output, as seen in the images. We obtained a much more accurate reading with less variation between measurements.
[image]

SilviaKarbid, 2024-09-25:
The new sensor code is uploded in the new code.

closed 2024-09-25 by SilviaKarbid