PubInv/moonrat #317: Lee, apply for an ORCID

RobertLRead, 2024-09-18:


ForrestErickson, 2024-09-18:
See: https://support.orcid.org/hc/en-us/articles/360006897454-How-do-I-register-for-an-ORCID-ID

ForrestErickson, 2025-09-05:
@melanielaporte 

I think this is it.
https://orcid.org/0009-0006-3624-0382

melanielaporte, 2025-09-05:
> think this is it. https: //orcid. org/0009-0006-3624-0382 — Reply to this
> because you were mentioned. Message
> ZjQcmQRYFpfptBannerStart
> This Message Is From an External Sender
> This message came from outside your organization.
>
> ZjQcmQRYFpfptBannerEnd
> <https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_PubInv_moonrat_issues_317-23issuecomment-2D3259161921&d=DwMCaQ&c=009klHSCxuh5AI1vNQzSO0KGjl4nbi2Q0M1QLJX9BeE&r=UDRyYbdt3uznO6lzmDlqX3dPjAUN2MbJJME6oFThrUE&m=0p7zk2okQydmgPcf0CCbfJoTXO4DWpS6gKPq3oeO6gB3VCGDM6aGufGokhlgsxNr&s=2bs2dIikddsfi1v16mx0ZQBWw9R0qc-gsNhkFQ3O1eg&e=>
>
> @melanielaporte
> <https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_melanielaporte&d=DwMCaQ&c=009klHSCxuh5AI1vNQzSO0KGjl4nbi2Q0M1QLJX9BeE&r=UDRyYbdt3uznO6lzmDlqX3dPjAUN2MbJJME6oFThrUE&m=0p7zk2okQydmgPcf0CCbfJoTXO4DWpS6gKPq3oeO6gB3VCGDM6aGufGokhlgsxNr&s=63GBsb2jkW5zCPCdxzOMS7bG212v9eWRe__767rUqUk&e=>
>
> I think this is it.
> https://orcid.org/0009-0006-3624-0382
> <https://urldefense.proofpoint.com/v2/url?u=https-3A__orcid.org_0009-2D0006-2D3624-2D0382&d=DwMCaQ&c=009klHSCxuh5AI1vNQzSO0KGjl4nbi2Q0M1QLJX9BeE&r=UDRyYbdt3uznO6lzmDlqX3dPjAUN2MbJJME6oFThrUE&m=0p7zk2okQydmgPcf0CCbfJoTXO4DWpS6gKPq3oeO6gB3VCGDM6aGufGokhlgsxNr&s=EDOAB2lzfgTTCDoqLcbKpSgmttiz2wpmVZZK3TLmJYU&e=>

Perfect.