PubInv/moonrat #318: Paper authorship: attach your professional portrait, biography, and affiliation

melanielaporte, 2024-09-18:
The following information is needed and will be published in the paper.
Please comment below.

Required:
- Full name to be listed on the paper
- One professional headshot (no body shots) – If you prefer a different photo than the one listed on the [Public Invention site](https://www.pubinv.org/inventors/), please upload it below. 
- Professional biography (100 words max)
- University affiliation

melanielaporte, 2024-09-18:
FULL NAME: Robert L. Read

AFFILIATION: Public Invention, 1709 Norris Dr., Austin, TX 78704

ORCID: 0000-0003-4749-7045

BIO: Robert L. Read has a BA in Computer Science from Rice University and PhD in Computer Science from the University of Texas at Austin. After holding director-level positions in software firms, he was a Presidential Innovation Fellow in 2013 and co-founded the 18F technology incubator in the US Federal Government. In 2018, he founded Public Invention, a US 501c3 public charity whose motto is “Invent in the public, for the Public.” Public Invention organizes teams of volunteers to work on free, open humanitarian research and engineering projects, recently including ferrofluid, tetrahedral robots, human medical ventilation, oxygen generation, and pure mathematics.

melanielaporte, 2024-09-18:
Silvia Aurora Casillas Romero is a PhD student in Electrical Engineering. Her current field of study is artificial intelligence, and she has a solid background in programming, frequency filter design, 3D modeling, and artificial intelligence. Silvia holds a Master's degree in Electrical Engineering, during which she worked on projects related to hydroponics. She is currently interested in deepening her knowledge of the development of artificial intelligence systems.
University aff: Cinvestav Unidad Guadalajara

melanielaporte, 2024-09-18:
L. Enrique Ruiz-Fernandez is an AI developer specializing in robotics. He is currently completing his Ph.D. in Robotics, holds a Master's degree in Electrical Engineering, and a Bachelor's degree in Control Engineering. His research focuses on autonomous navigation and collision avoidance for drones. With a strong background in control theory and electronics, he creates innovative solutions for complex engineering challenges. Additionally, he teaches key engineering courses, sharing his expertise with future engineers.
University aff: Cinvestav Unidad Guadalajara and Universidad de Guadalajara

melanielaporte, 2024-09-18:
(Forest) Lee Erickson
Public Inventor of the General Purpose Alarm Device (GPAD) Project
B.S. Physicist University of Minnesota.
I am the luckiest person to have ever lived because when I was ten years old, I was taken to the moon by Michael Collins, Buzz Aldrin and Neal Armstrong (as were about five billion other persons watching television.)

I have had a career doing elementary analog circuit design for audio, video and deflection systems and embedded system with some focus on electro- magnetic compatibility. I worked for a high volume consumer electronic company where I learned the importance of design for manufacturing and project management of medium- to large-multidisciplinary teams. I also ran my own company for 12 years where I had to take on almost all roles from time to time.

With Public Invention I have helped with the General Purpose Alarm Device (GPAD) project especially with Schematic Capture and PCB Design and testing. I wrote and simulated firmware for factory testing of the Version 1 PCB. From my Public Invention involvement, I have improved my understanding of how to use GitHub (and GitLab) for managing product development from my team mates. My short term interests are to document well the GPAD as a template to help Public Invention and others.

I am a refugee from the winters of Minnesota and live in East Tennessee where snow is sufficiently infrequent that it is mostly received as decorative when it happens. My supper power is NOVEL SPELLING ( and diverse spelling). My Github is found at: https://github.com/ForrestErickson

HJGV05, 2024-09-25:
Horacio Garc\'ia--V\'azquez} was born in Colima, M\'exico in 1995. He received the B. E. degree in Mechatronic Engineering  from the Colima Institute of Technology in 2018 and the M. Sc. degree in Electrical Engineering from the Center of Research and Advanced Studies (CINVESTAV-IPN) in 2020. He is currently pursuing the Ph. D. degree in Electrical Engineering at CINVESTAV-IPN.~His research interests include the output regulation problem, multi--agent systems, hybrid control systems and drones navigation.

[image]

melanielaporte, 2024-09-25:
Thank you @HJGV05!

closed 2024-10-30 by melanielaporte